coolsnowwolf/lede #3545: 今天编译固件失败，请大佬看下怎么解决

dl123100, 2020-03-04:
今天编译固件失败，之前一直正常的。似乎是wget-nossl和wget冲突，请大佬看下怎么解决。
2020-03-04T04:35:01.7297880Z Configuring luci-app-ddns.
2020-03-04T04:35:01.8777736Z Collected errors:
2020-03-04T04:35:01.8777889Z Configuring libuuid.
2020-03-04T04:35:01.8778312Z Configuring miniupnpd.
2020-03-04T04:35:01.8779802Z  * check_data_file_clashes: Package wget-nossl wants to install file /home/runner/work/lede/lede/build_dir/target-mipsel_24kc_musl/root-ramips/root/.wgetrc
2020-03-04T04:35:01.8780084Z Configuring kmod-usb3.
2020-03-04T04:35:01.8780418Z 	But that file is already provided by package  * wget
2020-03-04T04:35:01.8780815Z  * opkg_install_cmd: Cannot install package wget-nossl.
2020-03-04T04:35:01.8781079Z make[2]: *** [package/install] Error 255
2020-03-04T04:35:01.8783071Z Configuring coreutils-nohup.
2020-03-04T04:35:01.8783457Z Configuring ca-bundle.
2020-03-04T04:35:01.8783635Z Configuring libcurl4.
2020-03-04T04:35:01.8783811Z Configuring curl.
2020-03-04T04:35:01.8784115Z Configuring luci-app-openclash.
2020-03-04T04:35:01.8784314Z Configuring libsodium.
2020-03-04T04:35:01.8784625Z Configuring v2ray-plugin.
2020-03-04T04:35:01.8784960Z Configuring shadowsocksr-libev-alt.
2020-03-04T04:35:01.8785280Z Configuring iptables-mod-tproxy.
2020-03-04T04:35:01.8785497Z Configuring microsocks.
2020-03-04T04:35:01.8785808Z Configuring shadowsocks-libev-ss-redir.
2020-03-04T04:35:01.8786136Z Configuring kcptun-client.
2020-03-04T04:35:01.8786473Z Configuring shadowsocksr-libev-server.
2020-03-04T04:35:01.8786787Z Configuring luci-app-ssr-plus.
2020-03-04T04:35:01.8787118Z Configuring kmod-fs-nfs-common.
2020-03-04T04:35:01.8787447Z Configuring kmod-dnsresolver.
2020-03-04T04:35:01.8787771Z Configuring kmod-fs-nfs.
2020-03-04T04:35:01.8788100Z Configuring kmod-fs-nfs-v3.
2020-03-04T04:35:01.8788629Z Configuring kmod-fs-nfs-v4.
2020-03-04T04:35:01.8789221Z Configuring hostapd-common.
2020-03-04T04:35:01.8789548Z Configuring wpad-openssl.
2020-03-04T04:35:01.8789737Z Configuring odhcp6c.
2020-03-04T04:35:01.8789893Z Configuring unzip.
2020-03-04T04:35:01.8790086Z Configuring uci.
2020-03-04T04:35:01.8790386Z Configuring luci-proto-ipv6.
2020-03-04T04:35:01.8790561Z Configuring ip6tables.
2020-03-04T04:35:01.8790874Z Configuring kmod-ipt-nat6.
2020-03-04T04:35:01.8791226Z Configuring odhcpd-ipv6only.
2020-03-04T04:35:01.8791400Z Configuring ipv6helper.
2020-03-04T04:35:01.8791713Z Configuring wireless-regdb.
2020-03-04T04:35:01.8792039Z Configuring kmod-cfg80211.
2020-03-04T04:35:01.8792347Z Configuring kmod-mac80211.
2020-03-04T04:35:01.8792708Z Configuring kmod-mt76-core.
2020-03-04T04:35:01.8793023Z Configuring kmod-mt76x02-common.
2020-03-04T04:35:01.8793354Z Configuring kmod-mt76x2-common.
2020-03-04T04:35:01.8793682Z Configuring kmod-ipt-offload.
2020-03-04T04:35:01.8793988Z Configuring kmod-tcp-bbr.
2020-03-04T04:35:01.8794344Z Configuring luci-app-flowoffload.
2020-03-04T04:35:01.8794680Z Configuring luci-i18n-flowoffload-zh-cn.
2020-03-04T04:35:01.8794856Z Configuring dropbear.
2020-03-04T04:35:01.8795046Z Configuring tree.
2020-03-04T04:35:01.8795344Z Configuring luci-theme-argon.
2020-03-04T04:35:01.8795675Z Configuring kmod-crypto-authenc.
2020-03-04T04:35:01.8796001Z Configuring kmod-cryptodev.
2020-03-04T04:35:01.8796175Z Configuring mtd.
2020-03-04T04:35:01.8796593Z Configuring luci-i18n-vlmcsd-zh-cn.
2020-03-04T04:35:01.8796952Z Configuring luci-i18n-arpbind-zh-cn.
2020-03-04T04:35:01.8797287Z Configuring kmod-sched-cake.
2020-03-04T04:35:01.8797611Z Configuring sqm-scripts.
2020-03-04T04:35:01.8797922Z Configuring luci-app-sqm.
2020-03-04T04:35:01.8798279Z Configuring luci-i18n-aria2-zh-cn.
2020-03-04T04:35:01.8798590Z Configuring urandom-seed.
2020-03-04T04:35:01.8798921Z Configuring luci-i18n-qos-zh-cn.
2020-03-04T04:35:01.8799259Z Configuring libopenssl-devcrypto.
2020-03-04T04:35:01.8820947Z make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
2020-03-04T04:35:01.8821269Z Configuring kmod-crypto-arc4.
2020-03-04T04:35:01.8821715Z Configuring kmod-crypto-sha1.
2020-03-04T04:35:01.8822079Z Configuring kmod-crypto-ecb.
2020-03-04T04:35:01.8822412Z Configuring kmod-mppe.
2020-03-04T04:35:01.8822617Z Configuring ppp.
2020-03-04T04:35:01.8822951Z Configuring kmod-leds-gpio.
2020-03-04T04:35:01.8823320Z Configuring kmod-gpio-button-hotplug.
2020-03-04T04:35:01.8823509Z Configuring logd.
2020-03-04T04:35:01.8824413Z make: *** [world] Error 2
2020-03-04T04:35:01.8827155Z Configuring luci-app-accesscontrol.
2020-03-04T04:35:01.8827656Z Configuring luci-i18n-accesscontrol-zh-cn.
2020-03-04T04:35:01.8828010Z Configuring kmod-mt7603.
2020-03-04T04:35:01.8828508Z Configuring kmod-fs-ntfs.
2020-03-04T04:35:01.8828885Z Configuring luci-i18n-ddns-zh-cn.
2020-03-04T04:35:01.8829978Z Configuring smartdns.
2020-03-04T04:35:01.8830340Z Configuring luci-app-smartdns.
2020-03-04T04:35:01.8830735Z Configuring luci-i18n-smartdns-zh-cn.
2020-03-04T04:35:01.8831088Z Configuring simple-obfs.
2020-03-04T04:35:01.8831433Z Configuring kmod-nf-nathelper-extra.
2020-03-04T04:35:01.8831808Z Configuring luci-app-ramfree.
2020-03-04T04:35:01.8832158Z Configuring luci-i18n-ramfree-zh-cn.
2020-03-04T04:35:01.8832488Z Configuring luci-app-wol.
2020-03-04T04:35:01.8832864Z Configuring luci-i18n-wol-zh-cn.
2020-03-04T04:35:01.8833207Z Configuring luci-app-upnp.
2020-03-04T04:35:01.8833580Z Configuring luci-proto-relay.
2020-03-04T04:35:01.8833917Z Configuring default-settings.
2020-03-04T04:35:01.8834262Z Configuring luci-i18n-ttyd-zh-cn.
2020-03-04T04:35:01.8834486Z Configuring htop.
2020-03-04T04:35:01.8834814Z Configuring luci-i18n-base-zh-cn.
2020-03-04T04:35:01.8835169Z Configuring kmod-usb-ohci-pci.
2020-03-04T04:35:01.8835374Z Configuring tcping.
2020-03-04T04:35:01.8835700Z Configuring kmod-usb-ledtrig-usbport.
2020-03-04T04:35:01.8836088Z Configuring kmod-mt76x2.
2020-03-04T04:35:01.8836628Z Configuring luci-i18n-upnp-zh-cn.
2020-03-04T04:35:01.8836997Z Configuring libustream-openssl20150806.
2020-03-04T04:35:01.8837356Z Configuring kmod-fs-squashfs.
2020-03-04T04:35:01.8837549Z Configuring urngd.
2020-03-04T04:35:01.8837913Z Configuring luci-i18n-firewall-zh-cn.
2020-03-04T04:35:01.8838248Z Configuring ppp-mod-pppoe.
2020-03-04T04:35:01.8838604Z Configuring luci-app-passwall.
2020-03-04T04:35:01.8838975Z Configuring luci-i18n-nlbwmon-zh-cn.
2020-03-04T04:35:01.8839410Z package/Makefile:65: recipe for target 'package/install' failed
2020-03-04T04:35:01.8839828Z make[2]: Leaving directory '/home/runner/work/unknown/unknown'
2020-03-04T04:35:01.8840406Z package/Makefile:108: recipe for target '/home/runner/work/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
2020-03-04T04:35:01.8840784Z make[1]: Leaving directory '/home/runner/work/unknown/unknown'
2020-03-04T04:35:01.8841187Z /home/runner/work/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed

iyuangang, 2020-03-04:
1. `make menuconfig` 取消`wget-nossl` 安装.
2. `make downloads -j8 V=s` 把需求包都下载好
3. `make -j4 || make -j1 V=s` 如果出错，显示详细错误信息，再贴上来

dl123100, 2020-03-04:
> 1. `make menuconfig` 取消`wget-nossl` 安装.
> 2. `make downloads -j8 V=s` 把需求包都下载好
> 3. `make -j4 || make -j1 V=s` 如果出错，显示详细错误信息，再贴上来

能取消wget-nossl的话我就不发issue了。。。之前一直没问题的，就是今天更新后才遇到的。
本身是github actions编译的，make download看了log没问题的。

slitilanber, 2020-03-04:
wget 和wget-nossl冲突，只能选一项。建议只选wget

closed 2020-03-06 by coolsnowwolf