coolsnowwolf/lede #3546: 内网通过域名访问NAS失败

yzm157, 2020-03-04:
局域网无法用域名进入nas，外网可以。 #2916

Lean大，类似的问题有人提交过了，但是基本上都关闭了。
我的现象还有点不一样的地方。

我是自己编译的，并没有编译docker，所以应该不是docker的问题。
我是通过UPNP（2.0.20170421）进行的端口映射。
外网通过域名加端口的形式可以访问内网NAS。
内网不管是域名加端口还是WAN IP加端口都无法访问内网的NAS。

如果手动添加端口映射到NAS的IP，那么内网也是可以通过域名加端口的形式访问NAS的。

我在爱快和koolshare上，不手动映射端口，仅通过UPNP自动映射都是没有问题的。

烦请Lean大解惑下是什么原因造成的。

非常感谢！！！！

X64 华擎J3455 软路由

wspzf, 2020-03-04:
路由器加个host，把域名解析到192.168.1.1，应该可以解决你的问题，还免得绕一圈路

yzm157, 2020-03-05:
> 路由器加个host，把域名解析到192.168.1.1，应该可以解决你的问题，还免得绕一圈路

这个方式知道呀，我只是想知道原因，同时这也是个学习的过程。

hxchtiewei, 2020-03-05:
内网，dhcp里，自定义挟持域名设置下不就行了

closed 2020-03-06 by coolsnowwolf

xiaopinggao, 2022-08-19:
OpenWRT  lean版固件NAT 回环问题 #356

原因不是NAT loopback不起作用，而是bridge防火墙给阻挡了回流的流量，之前一直怀疑是这样，因为发现nmap一直显示filtered而不是closed 如已经编译可以在/etc/sysctl.conf 或 /etc/sysctl.d/11-br-netfilter.conf 修改

net.bridge.bridge-nf-call-arptables=0
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0

vi /etc/sysctl.d/11-br-netfilter.conf
net.bridge.bridge-nf-call-arptables=0
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0

按照上述修改
/etc/init.d/sysctl restart 重启搞定

参见：
https://github.com/klever1988/nanopi-openwrt/issues/356
https://github.com/coolsnowwolf/lede/issues/2103