coolsnowwolf/lede #3547: 系统启动跑一会码后黑屏

djs910, 2020-03-04:
反馈bug/问题模板，提建议请删除
xit
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 今天编译大神固件一直顺利，固件写入U盘后启动时跑一会吗就黑屏了，可以确定系统运行了。（盲键入reboot系统可以重新启动；远端可以通过浏览器进入路由器设定，功能都正常使用）
；映像文件转换城EXsi格式后在虚拟机下正常安装。

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

lylxoooooo, 2020-03-04:
我也遇到了这个问题，可能是显示驱动方面的原因，远程通过ssh是可以连接的

slitilanber, 2020-03-04:
我每次直接编译出来的都是这样黑屏。解决方案: make kernel_menuconfig，graphic drives 中添加了几项驱动搞定。但至今没搞清楚究竟是哪一项起的作用。求解。

lylxoooooo, 2020-03-04:
> 我每次直接编译出来的都是这样黑屏。解决方案: make kernel_menuconfig，graphic drives 中添加了几项驱动搞定。但至今没搞清楚究竟是哪一项起的作用。求解。
我尝试执行了一下make kernel_menuconfig，可是直接开始编译了，是正常情况吗？

lylxoooooo, 2020-03-04:
> 我每次直接编译出来的都是这样黑屏。解决方案: make kernel_menuconfig，graphic drives 中添加了几项驱动搞定。但至今没搞清楚究竟是哪一项起的作用。求解。

我执行了一下make kernel_menuconfig，可是直接开始编译了，是正常的吗

AmadeusGhost, 2020-03-04:
Refers to 1c6e4da87a1e20c0d810dc8dcbdf373764edd3d9
估计这个commit有问题

hxlls, 2020-03-04:
> Refers to [1c6e4da](https://github.com/coolsnowwolf/lede/commit/1c6e4da87a1e20c0d810dc8dcbdf373764edd3d9)
> 估计这个commit有问题

这个是加入了915核显的驱动，因为驱动了GPU 好像替代了tty 的显示，所以是这样的，实际上启动没问题，并且因为这个是个路由，所以显示无所谓，但是GPU的话有部分人有特殊用途，比如调用显卡配合docker进行转码转码之类的运算，.这个就看需求了 如果是需要这个启动显示就再内核把915的驱动去掉就行了，具体操作如下
[image]
进入Graphics support 》
[image]zhe这两个取消掉，就能解决
ps: 应该有其他的选项勾选后可以正常，
[image]比如这个，是启用传统的支持，等，但是实际个人上没测试，因为没这个需求，如果有人测试就更好了

djs910, 2020-03-05:
谢谢大家，有编译的问题互相交流，这样可以少走弯路

djs910, 2020-03-05:
这个选项在配置文件第几个菜单项呀

lylxoooooo, 2020-03-05:
这个问题已经解决了，Lean大已经revert 1c6e4da commit，现在已经没这个问题了。

djs910, 2020-03-05:
ok,谢谢。我重新编译一次。

closed 2020-03-05 by coolsnowwolf