coolsnowwolf/lede #3548: 最新版本ssr好像绕大陆ip模式有问题

zhuqiang007, 2020-03-04:
用gfw没问题，ip模式连不上

coolsnowwolf, 2020-03-04:
国内自定义的IP规则库崩了，换成下面这个官方地址

https://ispip.clang.cn/all_cn.txt

hcym, 2020-03-04:
我都是编译时替换好，省心

norikohxb, 2020-03-04:
我看lean大晚上又更新了lede，是不是晚上的版本就解决了这个问题了？

norikohxb, 2020-03-04:
这个崩了是什么意思，你说崩掉的那个txt，用网页浏览器打开也是一堆ip，这个崩了是指下面有些ip是坏掉的？

LisonFan, 2020-03-04:
> 国内自定义的IP规则库崩了，换成下面这个官方地址
> 
> https://ispip.clang.cn/all_cn.txt

用这个库也不行，绕过大陆 IP 也访问不了

norikohxb, 2020-03-04:
我是刚才又编译了一次，今天3次git pull， 今天晚上git pull以后好像好了。
感觉这个崩不知到是怎么定义，我用chrome去打开之前那个崩掉的一样是一堆ip的文本。
lean大能否赐教。
晚上这次的git pull好像解决了对吧

coolsnowwolf, 2020-03-04:
CIDR 的定义不同了

norikohxb, 2020-03-04:
lean大晚上的版本修复了对吧，我刚才又看到更新了，现在又编译了，今天编译4次了。
一早上起床编译一次。中午一次，吃晚饭一次，现在又一次。

LisonFan, 2020-03-04:
> 我是刚才又编译了一次，今天3次git pull， 今天晚上git pull以后好像好了。
> 感觉这个崩不知到是怎么定义，我用chrome去打开之前那个崩掉的一样是一堆ip的文本。
> lean大能否赐教。
> 晚上这次的git pull好像解决了对吧

> 这个崩了是什么意思，你说崩掉的那个txt，用网页浏览器打开也是一堆ip，这个崩了是指下面有些ip是坏掉的？

估计大部分人用的 IP 库是 IPIP 提供的那个，他们在今天更新库的时候没处理合并冲突就把库推上来了导致绕过大陆 IP 用不了

[image]

norikohxb, 2020-03-04:
lean的系统能否构建一个最基本的库（平时确定可以用稳定的）放在op本地，就是一些常规的网站不是很经常变动的。以本地文件的形式放在op，当出现问题的时候有一个安全模式可以调用。
比如说这个提供ip服务的网站因为地震天灾等情况。咱们的op不至于受到一个提供服务的网站影响。

hcym, 2020-03-04:
自己替换掉就结了，我都是替换掉，cn-ip只有50k不到，list更是好多一辈子不去的域名，什么功的域名大概有几十条，和cctv新华环球都在广告阻挡根域名的

coolsnowwolf, 2020-03-04:
本来这个就无需手动更新，内置了2个备用库，自动更新的没问题

mcucore, 2020-03-05:
自动更新时是否可以检查下获取的格式是否有效？

norikohxb, 2020-03-05:
对对，就是判断一下，更新之前的网站的ip是不是人为错误还是格式错误，如果错误就不更新了，这样路由器就更稳定了。
包括adbyby，wget最新的ip，这些，什么白名单，黑名单，这些东西，在更新前看能否检查一下是否是错误的，或者是导入错误格式以后出错能自动回退并且报错。

就好像在升级firmware之前先做一个被升级的文件有效性的检查。
虽然耗费lean大更多精力，但是这样做出来的路由器刀枪不入啊

closed 2020-03-05 by coolsnowwolf