coolsnowwolf/lede #3549: 新版SSR Plus+ Trojan节点添加的时候没有对应选项，使用socket5 SSR Plus+不能运行

nosixtools, 2020-03-04:
ShadowsocksR Plus+ 未运行

## 1.关于你要提交的问题
ShadowsocksR Plus+ 未运行 
## 2. 详细叙述
服务器节点类型：没有trojan，使用socket5添加后，不能正常运行，节点没有问题，PC使用客户端可以跑
### (2) 路由器型号和固件版本

A：极路由2 HC5761 

OpenWrt R20.2.15 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志

A：
2020-03-04 20:10:20 Socks5 REDIRECT/TPROXY, 1 Threads Started!
2020-03-04 20:09:39 Socks5 REDIRECT/TPROXY, 1 Threads Started!
2020-03-04 20:08:57 Socks5 REDIRECT/TPROXY, 1 Threads Started!
2020-03-04 20:08:14 Socks5 REDIRECT/TPROXY, 1 Threads Started!
2020-03-04 20:07:33 Socks5 REDIRECT/TPROXY, 1 Threads Started!
2020-03-04 20:06:50 Socks5 REDIRECT/TPROXY, 1 Threads Started!
2020-03-04 20:06:05 Socks5 REDIRECT/TPROXY, 1 Threads Started!

hcym, 2020-03-04:
🙄正在j13

closed 2020-03-04 by coolsnowwolf

nosixtools, 2020-03-04:
为啥给关了！

nosixtools, 2020-03-04:
新手编译失误了，已搞定，需要Luci->Applicaitions-> 选中 [] include Trojan