coolsnowwolf/lede #3550: 网易云解锁和ssr的启动顺序问题以及在ssr全局下网易云解锁无效

WROIATE, 2020-03-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

路由器重新启动，同时开启ssr和网易云解锁mini之后会出现网易云解锁mini失效的情况，查看防火墙发现是ssr的nat规则在网易云之前，重新启动网易云解锁将规则重新覆盖得以解决。但在ssr全局模式下网易云解锁mini即使重启仍无法生效。

### (2) 路由器型号和固件版本

newifi d2，OpenWrt R20.2.15 / LuCI Master (git-20.058.13156-2faf063)

### (3) 详细日志

下图为正常时的情况
[image]
下图为重启路由器后的情况
[image]

而在ssr全局下以上两种情况则皆不能生效

coolsnowwolf, 2020-03-04:
不要开全局

WROIATE, 2020-03-04:
> 不要开全局

因为在外需要进教育网内网所以开的全局，没有什么办法可以开全局的同时开吗？

coolsnowwolf, 2020-03-04:
没有

WROIATE, 2020-03-04:
> 没有

好的，谢谢lean