coolsnowwolf/lede #3551: 新增的SOCKS5代理无法连接 unsupported auth method

WellsWang, 2020-03-04:
本地用Trojan自带的Trojan-cli客户端建立SOCKS5代理，
使用Socks5代理功能连接本地未设置密码的SOCKS5代理，结果无法使用
查看本地SOCKS代理日志，有以下错误信息：
`[2020-03-04 21:34:16] [ERROR] 192.168.1.1:60244 unsupported auth method
[2020-03-04 21:34:16] [INFO] 192.168.1.1:60244 disconnected, 0 bytes received, 0 bytes sent, lasted for 0 seconds`

Trojan-cli的配置如下：
`{
    "run_type": "client",
    "local_addr": "0.0.0.0",
    "local_port": 1080,
    "remote_addr": "xxx.xxx.xxx",
    "remote_port": 443,
    "password": [
        "00000000"
    ],
    "log_level": 1,
    "ssl": {
        "verify": true,
        "verify_hostname": true,
        "cert": "fullchain.cer",
        "cipher_tls13":"TLS_AES_128_GCM_SHA256:TLS_CHACHA20_POLY1305_SHA256:TLS_AES_256_GCM_SHA384",
"sni": "",
        "alpn": [
            "h2",
            "http/1.1"
        ],
        "reuse_session": true,
        "session_ticket": false,
        "curves": ""
    },
    "tcp": {
        "no_delay": true,
        "keep_alive": true,
        "fast_open": false,
        "fast_open_qlen": 20
    }
}`

在其它电脑通过Chrome浏览器的SwitchyOmega插件配置192.168.1.137:1080 SOCKS代理是可以正常上网的。

请看下是何原因，谢谢！

martin0607, 2020-03-04:
刚准备发个issue就看到你的了，V2ray客户端也是一模一样的情况。以为是我自己编译错了，重新编了1次都一样。。。
[image]

martin0607, 2020-03-04:
在今天的潘多拉R20.3.4和19.07以及雕大LEDE上都一样，应该是插件问题

coolsnowwolf, 2020-03-04:
设个密码测试看看？

R3pl4c3r, 2020-03-04:
>`unsupported auth method`

意为不支持无认证模式，所以你需要设置个密码

coolsnowwolf, 2020-03-04:
[image]

新加了选项，作为无密码模式连接，可以测试看看

WellsWang, 2020-03-05:
@coolsnowwolf 感谢，使用无密码模式后解决问题，谢谢！
Close Issue

closed 2020-03-05 by WellsWang