coolsnowwolf/lede #3558: dockerman更新后，没有网络

jy3344511, 2020-03-04:
之前使用dockerman，创建容器后默认使用的bridge网络，qBittorrent可以正常下载。更新后，发现qBittorrent无法下载了，在已经创建的容器中，怎么修改网络为host？尝试重新创建容器，在命令中，添加--net=host，但是提示无法创建容器。

用过iptables -t nat -A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE、ip link del docker0
reboot，都没有用。

yuban10703, 2020-03-04:
https://github.com/coolsnowwolf/lede/issues/3515

jy3344511, 2020-03-04:
> #3515

执行rm -rf /var/lib/docker && service dockerd restart后，提示uci: Entry not found
sh: true: unknown operand

yuban10703, 2020-03-04:
> > #3515
> 
> 你发的那个，我看过了，怎么做？

我试了,执行那个就好了,不行的话去开一下wan口访问,再执行一下,emmmmm

jy3344511, 2020-03-05:
> > > #3515
> > 
> > 
> > 你发的那个，我看过了，怎么做？
> 
> 我试了,执行那个就好了,不行的话去开一下wan口访问,再执行一下,emmmmm

就是执行这个啊iptables -t nat -A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE、ip link del docker0，执行了，没用啊。

另外，旁路由，怎么开wan口？

wspzf, 2020-03-05:
> > > #3515
> > 
> > 
> > 你发的那个，我看过了，怎么做？
> 
> 我试了,执行那个就好了,不行的话去开一下wan口访问,再执行一下,emmmmm

碰到同样问题,打开WAN口执行,重启就好了,谢谢

wspzf, 2020-03-05:
> > > #3515
> > 
> > 
> > 你发的那个，我看过了，怎么做？
> 
> 我试了,执行那个就好了,不行的话去开一下wan口访问,再执行一下,emmmmm

重启后又不通了.....

YYiiEt, 2020-03-05:
可以试试 #2872 的提供的方法，我目前使用的基于lean的源码旁路由就是通过该方法解决的。

jy3344511, 2020-03-05:
> 到同样问题,打开WAN口执行,重启就好了,谢谢

你的意思是，添加一个新的接口为wan口，然后执行iptables -t nat -A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE、ip link del docker0这两个吗？

jy3344511, 2020-03-05:
> 可以试试 #2872 的提供的方法，我目前使用的基于lean的源码旁路由就是通过该方法解决的。

解决方案：
然后vim /etc/sysctl.conf，添加下列内容：
net.bridge.bridge-nf-call-ip6tables = 1
net.bridge.bridge-nf-call-iptables = 1
net.bridge.bridge-nf-call-arptables = 1
后台修改设置：
Luci>网络>防火墙>转发：接受
Luci>状态>防火墙>重启防火墙
最后ssh执行service dockerd restart

按照这个做了，没用啊，到底怎么弄啊？

yuban10703, 2020-03-05:
> > > > #3515
> > > 
> > > 
> > > 你发的那个，我看过了，怎么做？
> > 
> > 
> > 我试了,执行那个就好了,不行的话去开一下wan口访问,再执行一下,emmmmm
> 
> 再次重启后又不通了.....

我的删一次后貌似就正常了,重启也没问题...

jy3344511, 2020-03-05:
> 删一次后貌似就正常了

你具体的操作，能发一下吗？

yuban10703, 2020-03-05:
> > 删一次后貌似就正常了
> 
> 你具体的操作，能发一下吗？

[image]
勾上wan访问
执行rm -rf /var/lib/docker && service dockerd restart

jy3344511, 2020-03-05:
> > > 删一次后貌似就正常了
> > 
> > 
> > 你具体的操作，能发一下吗？
> 
> [image]
> 勾上wan访问
> 执行rm -rf /var/lib/docker && service dockerd restart

多谢。

closed 2020-03-05 by jy3344511