coolsnowwolf/lede #3560: R6850编译错误

edboffical, 2020-03-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
想编译R6850的固件 选了v2ray和trojan 编译错误
### (1) 具体问题

A：

一直编译错误

### (2) 路由器型号和固件版本

A：

网件R6850 target-mipsel_24kc

### (3) 详细日志

A：admin@iZj6c4zttul0nhhy4xsx6yZ:~/lede$ make -j1 V=s
time: target/linux/prereq#0.24#0.00#0.45
make[1]: Entering directory '/home/admin/lede'
make[2]: Entering directory '/home/admin/lede'
+ mkdir -p /home/admin/lede/staging_dir/target-mipsel_24kc_musl
+ cd /home/admin/lede/staging_dir/target-mipsel_24kc_musl
+ mkdir -p bin lib include stamp
mkdir -p /home/admin/lede/build_dir/target-mipsel_24kc_musl/stamp
touch /home/admin/lede/staging_dir/target-mipsel_24kc_musl/.prepared
+ mkdir -p /home/admin/lede/staging_dir/host
+ cd /home/admin/lede/staging_dir/host
+ mkdir -p bin lib include stamp
mkdir -p /home/admin/lede/build_dir/host/stamp /home/admin/lede/staging_dir/host/include/sys
install -m0644 /home/admin/lede/tools/include/*.h /home/admin/lede/staging_dir/host/include/
install -m0644 /home/admin/lede/tools/include/sys/*.h /home/admin/lede/staging_dir/host/include/sys/
ln -sf lib /home/admin/lede/staging_dir/host/lib64
touch /home/admin/lede/staging_dir/host/.prepared
make[3]: Entering directory '/home/admin/lede/tools/flock'
make[3]: Leaving directory '/home/admin/lede/tools/flock'
time: tools/flock/compile#0.00#0.00#0.04
make[3]: Entering directory '/home/admin/lede/tools/xz'
make[3]: Leaving directory '/home/admin/lede/tools/xz'
time: tools/xz/compile#0.00#0.00#0.09
make[3]: Entering directory '/home/admin/lede/tools/sed'
make[3]: Leaving directory '/home/admin/lede/tools/sed'
time: tools/sed/compile#0.00#0.00#0.07
make[3]: Entering directory '/home/admin/lede/tools/patch'
make[3]: Leaving directory '/home/admin/lede/tools/patch'
time: tools/patch/compile#0.00#0.00#0.06
make[3]: Entering directory '/home/admin/lede/tools/tar'
make[3]: Leaving directory '/home/admin/lede/tools/tar'
time: tools/tar/compile#0.00#0.00#0.08
make[3]: Entering directory '/home/admin/lede/tools/m4'
make[3]: Leaving directory '/home/admin/lede/tools/m4'
time: tools/m4/compile#0.00#0.00#0.08
make[3]: Entering directory '/home/admin/lede/tools/autoconf'
make[3]: Leaving directory '/home/admin/lede/tools/autoconf'
time: tools/autoconf/compile#0.00#0.00#0.06
make[3]: Entering directory '/home/admin/lede/tools/pkg-config'
make[3]: Leaving directory '/home/admin/lede/tools/pkg-config'
time: tools/pkg-config/compile#0.00#0.00#0.07
make[3]: Entering directory '/home/admin/lede/tools/automake'
make[3]: Leaving directory '/home/admin/lede/tools/automake'
time: tools/automake/compile#0.01#0.00#0.09
make[3]: Entering directory '/home/admin/lede/tools/missing-macros'
make[3]: Leaving directory '/home/admin/lede/tools/missing-macros'
time: tools/missing-macros/compile#0.00#0.00#0.03
[... 154 lines cut ...]
make[5]: Entering directory '/home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0-initial/fixincludes'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0-initial/fixincludes'
make[5]: Entering directory '/home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0-initial/gcc'
build/genautomata /home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0/gcc/common.md /home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0/gcc/config/mips/mips.md \
  insn-conditions.md > tmp-automata.c
/bin/bash: line 1: 14614 Killed                  build/genautomata /home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0/gcc/common.md /home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0/gcc/config/mips/mips.md insn-conditions.md > tmp-automata.c
Makefile:2278: recipe for target 's-automata' failed
make[5]: *** [s-automata] Error 137
make[5]: Leaving directory '/home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0-initial/gcc'
Makefile:4210: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0-initial'
Makefile:36: recipe for target '/home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0-initial/.built' failed
make[3]: *** [/home/admin/lede/build_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/gcc-7.5.0-initial/.built] Error 2
make[3]: Leaving directory '/home/admin/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#4.54#0.42#6.00
toolchain/Makefile:98: recipe for target 'toolchain/gcc/initial/compile' failed
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/admin/lede'
toolchain/Makefile:96: recipe for target '/home/admin/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/admin/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/admin/lede'
/home/admin/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

FoxDring, 2020-03-04:
类似的错误

tmpaxx, 2020-03-04:
erx同样的错误
/home/admin/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

edboffical, 2020-03-05:
> erx同样的错误
> /home/admin/lede/include/toplevel.mk:225: recipe for target 'world' failed
> make: *** [world] Error 2

你们找到问题了吗 我换了个机器编译还是错误

closed 2020-03-05 by coolsnowwolf

edboffical, 2020-03-06:
用github actions去编译就不会出错了 应该是编译环境的问题

tmpaxx, 2020-03-16:
我找到问题了，因为我用的WIN10 UBUNTU的APP编译的，有个文件夹的目录会导致编译失败，用虚拟机就可以了。