coolsnowwolf/lede #3570: 问个问题，adbybyplus  和 ssrplus 内的广告屏蔽会冲突吗？

keivenliao, 2020-03-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：问个问题，adbybyplus  和 ssrplus 内的广告屏蔽会冲突吗？还是只保留其中一个即可？

### (2) 路由器型号和固件版本

A：X86 2020-3-5 11:10 分编译版本

### (3) 详细日志

A：无

hyird, 2020-03-05:
不会

iyuangang, 2020-03-05:
没有冲突，广告屏蔽规则相当于逻辑与的关系。内存会多占用10M左右，基本可以忽略影响。
只开启adbybyplus 的内存占用。
[image]

同时开启adbybyplus 和ssl plus + 广告屏蔽后
[image]

Wilson-coder, 2020-03-05:
> 没有冲突，广告屏蔽规则相当于逻辑与的关系。内存会多占用10M左右，基本可以忽略影响。
> 只开启adbybyplus 的内存占用。
> [image]
> 
> 同时开启adbybyplus 和ssl plus + 广告屏蔽后
> [image]

我两个模块都是打开状态，不过用电脑播放优酷视频，仍然需要每个视频开始等待几十秒的广告。好像是没有效果，是不是设置有哪里不对头？

closed 2020-03-05 by coolsnowwolf