coolsnowwolf/lede #3573: 关于螃蟹8168网卡驱动的问题

hgygyn, 2020-03-05:
我使用的软路由机子是HTPC版的小型电脑，自带两个螃蟹的8168网卡，可是在源码里只提供8169的网卡驱动，能请你把8168的网卡驱动编译进去吗？感谢你的辛苦付出

hxlls, 2020-03-05:
在提issues 时候就不能先自己测试么？连装都没装，就说没有8168驱动？

Her0R0cket, 2020-03-05:
驱动名不是一一对应的,大多是通用驱动

gpeditmsc1986, 2020-03-05:
这里有，加进去就行了，我就这么干的
https://github.com/BROBIRD/openwrt-r8168

hgygyn, 2020-03-05:
我是希望你能直接添加进去，而不需要自己添加
我是一枚真正的小白，虽然有自己编译成功的案列，但是像这种添加驱动的活还没自己弄过，编译用的文件里好多东西都不知道是做什么用的

 https://github.com/BROBIRD/openwrt-r8168

dolphinpaopao, 2020-03-05:
你编译的固件8168目前无法驱动吗？我自编译的8111e好像和8168一个驱动是可以正常用8169驱动的。没有测试就提出问题是不负责的。请预先测试并贴出详细相关内容。学习是需要成本，别人的知识也不是天上掉下来的。

hgygyn, 2020-03-05:
很抱歉给你带来了麻烦，

ckxt, 2020-03-05:
rtl的81xx系列千兆网卡驱动好像是通用的，你实际测试过了吗

hgygyn, 2020-03-05:
有在使用源代码里带的8169驱动，就是开机后要有大概1分钟才能获取到IP

dolphinpaopao, 2020-03-05:
官网是有更新，今年一月八号，内核4.15，可以试试gpeditmsc1986给的方式，自我学习尝试一下。
https://www.realtek.com/zh-tw/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software

Her0R0cket, 2020-03-05:
> 
> 有在使用源代码里带的8169驱动，就是开机后要有大概1分钟才能获取到IP
> […](#)

我rtl8211的网卡也是8169驱动,系统运行15秒后识别了硬件,但联网好像是25秒后

punkzhang, 2020-03-06:
其实我好像也有这个问题。8168/8111有专门的驱动。
每次我自己添加之后在menuconfig里总是找不到。
然后就勉强用8169的代替了。

closed 2020-03-06 by coolsnowwolf

shikkoku, 2020-06-16:
https://github.com/goldkeyber112/openwrt-r8168/releases/download/8.048.03/kmod-r8168_4.19.122+8.048.03-x86-3_x86_64.ipk

上传到路由后opkg安装，然后把kmod-r8169驱动remove了，然后重启你的路由，应该就能用上官方的驱动了。注意内核版本。

shikkoku, 2020-06-19:
fixed in https://github.com/coolsnowwolf/lede/commit/ba8a697b767908a4397e11241b5c394bd29f2d21