coolsnowwolf/lede #3577: qBittorrent webui无启动页面

EstFoxs, 2020-03-05:
固件版本20200305 路由型号86_64
没有按键，通过ip也无法访问
[image]
Transmission 没有这个问题

bin20088, 2020-03-05:
并不存在。一切正常X86

EstFoxs, 2020-03-05:
那奇怪了，晚点重新刷机

closed 2020-03-05 by EstFoxs

EstFoxs, 2020-03-05:
> 并不存在。一切正常X86

确定正常？刚才重刷了，也不行啊

bin20088, 2020-03-05:
QQ群里我传的固件，随便测。这玩意还糊你？
[image]

bin20088, 2020-03-05:
[image]
[image]

EstFoxs, 2020-03-05:
重新拉包了，已经在编译，一会测试

EstFoxs, 2020-03-05:
不行，我还了三台软路由都这个问题，你内核多少

bin20088, 2020-03-05:
@EstFoxs 106