coolsnowwolf/lede #3578: 竟斗云又开始刷日志了蛋疼

stock169, 2020-03-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
以前是刷IPV6不支持后来好了，现在新版又来了估计还是那个老问题

### (2) 路由器型号和固件版本

A：
OpenWrt R20.2.15 /LuCI Master (git-20.062.50551-a6189c5) 

### (3) 详细日志

A：[   26.503853] device eth0 entered promiscuous mode
[   26.505953] br-lan: port 1(eth0.1) entered blocking state
[   26.510115] br-lan: port 1(eth0.1) entered disabled state
[   26.515823] device eth0.1 entered promiscuous mode
[   26.528652] br-lan: port 1(eth0.1) entered blocking state
[   26.528707] br-lan: port 1(eth0.1) entered forwarding state
[   26.533372] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   27.702482] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   27.703665] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   30.887013] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   32.517440] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   32.520993] br-lan: port 2(wlan1) entered blocking state
[   32.522182] br-lan: port 2(wlan1) entered disabled state
[   32.528166] device wlan1 entered promiscuous mode
[   32.533218] br-lan: port 2(wlan1) entered blocking state
[   32.537633] br-lan: port 2(wlan1) entered forwarding state
[   32.544029] br-lan: port 2(wlan1) entered disabled state
[   32.548837] br-lan: port 3(wlan0) entered blocking state
[   32.553916] br-lan: port 3(wlan0) entered disabled state
[   32.559426] device wlan0 entered promiscuous mode
[   33.023073] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.023308] br-lan: port 3(wlan0) entered blocking state
[   33.028509] br-lan: port 3(wlan0) entered forwarding state
[   33.308942] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   33.309214] br-lan: port 2(wlan1) entered blocking state
[   33.314440] br-lan: port 2(wlan1) entered forwarding state
[ 1275.870014] mmc0: Reset 0x1 never completed.
[10606.406905] ess_edma c080000.edma: 
[10627.154411] ess_edma c080000.edma: 
[10627.845720] ess_edma c080000.edma: 
[10633.058532] ess_edma c080000.edma: 
[10633.123317] ess_edma c080000.edma: 
[10637.009054] ess_edma c080000.edma: 
[10637.264245] ess_edma c080000.edma: 
[10637.602600] ess_edma c080000.edma: 
[10641.620548] ess_edma c080000.edma: 
[10646.141013] ess_edma c080000.edma: 
[10658.852846] ess_edma c080000.edma: 
[10660.903674] ess_edma c080000.edma: 
[10662.454894] ess_edma c080000.edma: 
[10663.277700] ess_edma c080000.edma: 
[10670.945550] ess_edma c080000.edma: 
[10674.731144] ess_edma c080000.edma: 
[10675.293061] ess_edma c080000.edma: 
[10684.637797] ess_edma c080000.edma: 
[10684.861361] ess_edma c080000.edma: 
[10686.449086] ess_edma c080000.edma: 
[10688.454486] ess_edma c080000.edma: 
[10689.248305] ess_edma c080000.edma: 
[10689.346071] ess_edma c080000.edma: 
[10693.574157] ess_edma c080000.edma: 
[10694.015723] ess_edma c080000.edma: 
[10696.420521] ess_edma c080000.edma: 
[10697.371367] ess_edma c080000.edma: 
[10706.424522] ess_edma c080000.edma: 
[10707.276822] ess_edma c080000.edma: 
[10707.821619] ess_edma c080000.edma: 
[10708.130983] ess_edma c080000.edma: 
[10709.215206] ess_edma c080000.edma: 
[10710.522626] ess_edma c080000.edma: 
[10710.735317] ess_edma c080000.edma: 
[10711.089164] ess_edma c080000.edma: 
[10711.092911] ess_edma c080000.edma: 
[10711.576309] ess_edma c080000.edma: 
[10711.632035] ess_edma c080000.edma: 
[10711.639530] ess_edma c080000.edma: 
[10712.948278] ess_edma c080000.edma: 
[10713.750673] ess_edma c080000.edma: 
[10714.363546] ess_edma c080000.edma: 
[10715.061385] ess_edma c080000.edma: 
[10715.512833] ess_edma c080000.edma: 
[10721.221596] ess_edma c080000.edma: 
[10721.932501] ess_edma c080000.edma: 
[10723.792738] ess_edma c080000.edma: 
[10724.802402] ess_edma c080000.edma: 
[10727.008250] ess_edma c080000.edma: 
[10728.039334] ess_edma c080000.edma: 
[10728.297192] ess_edma c080000.edma: 
[10729.800662] ess_edma c080000.edma: 
[10730.531772] ess_edma c080000.edma:

coolsnowwolf, 2020-03-05:
哈哈哈 ，又是 ipv6

coolsnowwolf, 2020-03-05:
修复了

closed 2020-03-05 by coolsnowwolf