coolsnowwolf/lede #3579: openvpn server 多客户端支持疑问

ckxt, 2020-03-05:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：启用了OpenVPN Server功能之后，手机上连接成功，获得的IP是10.8.0.2,
      同时用笔记本电脑测试，发现获得的IP还是10.8.0.2，过了不久手机端的就提示连接失败，

### (2) 路由器型号和固件版本

A：X86软路由  OpenWrt R20.2.15 / LuCI Master (git-20.058.13156-2faf063)

 是OpenVPN server 只支持一个设备同时在线吗？

谢谢

hxlls, 2020-03-05:
这个可以到openvpn的上游去提，这样比较合适

Biggulu, 2020-03-05:
大雕内置的证书只有一个，想要多客户端同时连接，请自签证书即可

segadrive, 2020-03-05:
希望能支持ipv6

ckxt, 2020-03-05:
> 大雕内置的证书只有一个，想要多客户端同时连接，请自签证书即可

大神能不能指个路  怎么自签证书..我玩过自己在ubuntu上创建openvpn server, 但是对这个openwrt里的不熟悉 不知道CA证书这些放在什么地方

ckxt, 2020-03-06:
> 大雕内置的证书只有一个，想要多客户端同时连接，请自签证书即可

找到了 谢谢

`root@OpenWrt:~# cd /etc/openvpn/
root@OpenWrt:/etc/openvpn# ls
ca.crt       client1.crt  client1.key  dh1024.pem   server.crt   server.key
root@OpenWrt:/etc/openvpn#
`

closed 2020-03-06 by ckxt