coolsnowwolf/lede #3581: openwrt--pppoe拨号能拨到外网ip，不能上网，要在防火墙-自定义规则里 重启防火墙 就可以上网。4.14.171

0533qz, 2020-03-05:
1.openwrt--pppoe拨号能拨到外网ip，不能上网，要在防火墙-自定义规则里 重启防火墙 就可以上网。4.14.171
重启路由还是这样， 能拨到外网ip， 不能上网， 要重启防火墙。

 (2) 竞斗云、新路由3-固件版本4.14.171

 (3) 共性问题

0533qz, 2020-03-05:
动态ip没问题。

0533qz, 2020-03-05:
主机名 | OpenWrt
-- | --
主机型号 | P&W R619AC 128M
架构 | ARMv7 Processor rev 5 (v7l) x 4
固件版本 | OpenWrt R20.2.15 / LuCI Master (git-20.062.50551-a6189c5)
内核版本 | 4.14.171
本地时间 | Thu Mar 5 16:19:45 2020
运行时间 | 1h 13m 20s
平均负载 | 0.25, 0.20, 0.18
CPU 使用率（%） | 1.6%

kawon7317, 2020-03-05:
是的，我的树莓派4也是这样，不过很少重启，就没管了

coolsnowwolf, 2020-03-05:
应该是有个程序 hook 住了防火墙，你们默认编译试试吧

0533qz, 2020-03-05:
> 应该是有个程序 hook 住了防火墙，你们默认编译试试吧

动态ip没问题，
pppoe，获取到ip，如果不能上网，到防火墙-自定义规则  按下 重启防火墙 选项。 就可以了。
没什么大问题， 就是断电重启什么的进去重启防火墙。
一样用，一样上网。

不知道的， pppoe可能就衰了。 重启防火墙。

0533qz, 2020-03-05:
> 是的，我的树莓派4也是这样，不过很少重启，就没管了

动态ip没问题，
pppoe，获取到ip，如果不能上网，到防火墙-自定义规则  按下 重启防火墙 选项。 就可以了。
没什么大问题， 就是断电重启什么的进去重启防火墙。
一样用，一样上网。

不知道的， pppoe可能就衰了。 重启防火墙。

joketans, 2020-03-06:
这几天我每天都编译安装过，光猫桥接，软路由pppoe拨号一切正常，没这问题。

closed 2020-03-06 by coolsnowwolf

cbdog94, 2020-03-08:
+1 也有这种问题 4.14.171 华硕arch17

PussAzuki, 2020-04-01:
> 应该是有个程序 hook 住了防火墙，你们默认编译试试吧

老大你这位于 package/network/config/firewall/patches的fullconenat.patch已经歪了，defaults.c 的应该打在 -46,7 +46,9
我测试了下，改了之后就修好了这个问题

Erope, 2020-07-04:
的确 我也有这个问题
看了看重启前后的iptables的FORWARD链，差距挺大的，应该是哪个程序hook了防火墙