coolsnowwolf/lede #3582: 支持ERX的硬件加密吗?

crabhit, 2020-03-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ERX上加密算法是通过CPU来计算的 有没有offload给硬件?

### (2) 路由器型号和固件版本

A：Ubnt ERX, OpenWrt R20.1.24 / LuCI Master (git-20.045.63360-69eac90)

### (3) 详细日志

A：

ckxt, 2020-03-05:
联发科MT7261 肯定有了

crabhit, 2020-03-06:
> 联发科MT7261 肯定有了

怎么启用硬件加密呢? 默认应该是没有启用的。

ckxt, 2020-03-06:
> > 联发科MT7261 肯定有了
> 
> 怎么启用硬件加密呢? 默认应该是没有启用的。

嘿嘿 我没看你的标题，我看到了offload就回复你了

OpenWRT里有这几个 你如果自己玩了OpenWRT的话 找到以下不难的：

HWNAT 启用硬件HWNAT加速（依赖特定的硬件，例如 MTK 762x 系列）

 开启 Flow Offloading 转发加速. (降低CPU占用 / 增强路由转发性能，和多WAN或者QOS功能可能会冲突)

crabhit, 2020-03-06:
> > > 联发科MT7261 肯定有了
> > 
> > 
> > 怎么启用硬件加密呢? 默认应该是没有启用的。
> 
> 嘿嘿 我没看你的标题，我看到了offload就回复你了
> 
> OpenWRT里有这几个 你如果自己玩了OpenWRT的话 找到以下不难的：
> 
> HWNAT 启用硬件HWNAT加速（依赖特定的硬件，例如 MTK 762x 系列）
> 
> 开启 Flow Offloading 转发加速. (降低CPU占用 / 增强路由转发性能，和多WAN或者QOS功能可能会冲突)

哦 你说的是转发的offload, 这个是有的。

我想问的是MT7621应该硬件也支持加密运算的offload 算SHA加密之类的。怎么利用好这块的硬件。

ckxt, 2020-03-06:
原来你说的是AES，那你先去查一下MT7621到底支不支持AES指令集，而不是在这里问。
如果MT7621带AES指令集，那加解密肯定会有offload了，本身不支持AES指令集的话，谈什么offload呢

closed 2020-03-06 by coolsnowwolf