coolsnowwolf/lede #3583: 一个关于issues的问题

iKenzo, 2020-03-05:
想问各位，为什我我的issue总是会被关掉？这里有什么rules需要follow 吗？我在哪里可以找到呢？

[image]

dolphinpaopao, 2020-03-05:
issue的含义不是用来讨论这种能否选择节点问题的，这类问题大多数可以通过预先搜索或者去论坛QQ群等方式获得答案。
因为有人不胜其烦的重复开issue询问是否能开socks服务之类，所以作者直接默认了。解决大家问题。其实你搜一下应该也能看到的。

iKenzo, 2020-03-05:
哦，谢谢！我没有qq，也不知到去哪个论坛问，只知道这里，所以才来问，有telegram的group吗？或者论坛的链接？非常感谢！

dolphinpaopao, 2020-03-05:
> 哦，谢谢！我没有qq，也不知到去哪个论坛问，只知道这里，所以才来问，有telegram的group吗？或者论坛的链接？非常感谢！

仔细多看看，本项目readme里面就有。

crabhit, 2020-03-05:
> > 哦，谢谢！我没有qq，也不知到去哪个论坛问，只知道这里，所以才来问，有telegram的group吗？或者论坛的链接？非常感谢！
> 
> 仔细多看看，本项目readme里面就有。

QQ群还是没有电报群方便

coolsnowwolf, 2020-03-05:
issue 不是版聊，这里不是论坛

iKenzo, 2020-03-05:
感谢各位百忙中的回复，谢谢！
因为我平时不怎么用中文的论坛，所以壮着胆子再问一次，如果不想用QQ的话，哪里是靠谱的“版聊”的地方？或者official的论坛？我Google了一下眼花缭乱，太多不同的论坛，年纪大了看得头都疼！麻烦各位了！

dolphinpaopao, 2020-03-05:
> 感谢各位百忙中的回复，谢谢！
> 因为我平时不怎么用中文的论坛，所以壮着胆子再问一次，如果不想用QQ的话，哪里是靠谱的“版聊”的地方？或者official的论坛？我Google了一下眼花缭乱，太多不同的论坛，年纪大了看得头都疼！麻烦各位了！

非中文的话，可能要去op官方论坛讨论了。毕竟本项目的很多功能比较特色化。

iKenzo, 2020-03-05:
> > 感谢各位百忙中的回复，谢谢！
> > 因为我平时不怎么用中文的论坛，所以壮着胆子再问一次，如果不想用QQ的话，哪里是靠谱的“版聊”的地方？或者official的论坛？我Google了一下眼花缭乱，太多不同的论坛，年纪大了看得头都疼！麻烦各位了！
> 
> 非中文的话，可能要去op官方论坛讨论了。毕竟本项目的很多功能比较特色化。

是的，可能是我没有表达清楚，我就是想麻烦各位推荐中文的靠谱论坛

R3pl4c3r, 2020-03-06:
> > > 感谢各位百忙中的回复，谢谢！
> > > 因为我平时不怎么用中文的论坛，所以壮着胆子再问一次，如果不想用QQ的话，哪里是靠谱的“版聊”的地方？或者official的论坛？我Google了一下眼花缭乱，太多不同的论坛，年纪大了看得头都疼！麻烦各位了！
> > 
> > 
> > 非中文的话，可能要去op官方论坛讨论了。毕竟本项目的很多功能比较特色化。
> 
> 是的，可能是我没有表达清楚，我就是想麻烦各位推荐中文的靠谱论坛

恩山

closed 2020-03-06 by coolsnowwolf

iKenzo, 2020-03-06:
> > > > 感谢各位百忙中的回复，谢谢！
> > > > 因为我平时不怎么用中文的论坛，所以壮着胆子再问一次，如果不想用QQ的话，哪里是靠谱的“版聊”的地方？或者official的论坛？我Google了一下眼花缭乱，太多不同的论坛，年纪大了看得头都疼！麻烦各位了！
> > > 
> > > 
> > > 非中文的话，可能要去op官方论坛讨论了。毕竟本项目的很多功能比较特色化。
> > 
> > 
> > 是的，可能是我没有表达清楚，我就是想麻烦各位推荐中文的靠谱论坛
> 
> 恩山

非常感谢！！