coolsnowwolf/lede #3596: 能否增加 指定域名 走指定的 XXR Plus 线路，比如奈飞专用线路，常用上网专用线路

jerryrat, 2020-03-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：能否增加 指定域名 走指定的 XXR Plus 线路，比如奈飞专用线路，常用上网专用线路

### (2) 路由器型号和固件版本

A：X86 R7500V2 R7800

### (3) 详细日志

A：无

yangjiey2k, 2020-03-06:
其实我也有这个需求

martin0607, 2020-03-06:
那又是另一个模式了，可以去用Clash，类似Surge的分流

jerryrat, 2020-03-06:
> 其实我也有这个需求

我解决的办法是 2级路由，一级路由SSR+只添加 奈飞线路，二级路由SSR+添加其他机场线路，二级路由"不走代理的域名" 增加 

nefxvideo.net
netflix.com

jerryrat, 2020-03-06:
> 其实我也有这个需求

我试过这样：2级路由，一级SSR+只添加 奈飞机场，二级路由ssr+只添加普通机场，二级路由 “不走代理的域名” 里面添加

Netflix.com
nflxvideo.net

然后 上述路由 走的 是1级路由，其他走的二级路由的SSR+

只是测试过，不知道二级路由的SSR+线路有没有 再次经过一级的 ssr+线路。

coolsnowwolf, 2020-03-06:
很难做到，因为在群里号召了很久没人愿意借奈飞账号给我测试

haig233, 2020-03-06:
> 很难做到，因为在群里号召了很久没人愿意借奈飞账号给我测试

大雕给个邮箱，我可以借我的号给你

closed 2020-03-06 by coolsnowwolf

abclai, 2020-03-06:
> 很难做到，因为在群里号召了很久没人愿意借奈飞账号给我测试
能不能邮箱发给我，我赞助一个月的奈飞会员给你测试啦