coolsnowwolf/lede #3597: Configure my own anti DNS pollution server

Limezy, 2020-03-06:
Hi, in the ShadowSocksR Plus+ Settings menu, I would like to setup an anti DNS pollution server that is not in that list (for example CleanBrowsing DNS)

[image]

How can I do that ?
Or can I setup the V2Ray client to use my V2Ray server DNS resolution ?
Thanks !

jerryrat, 2020-03-06:
hi,  in SSR+, you can input  customization DNS
[image]

iyuangang, 2020-03-06:
> Hi, in the ShadowSocksR Plus+ Settings menu, I would like to setup an anti DNS pollution server that is not in that list (for example CleanBrowsing DNS)
> 
> [image]
> 
> How can I do that ?
> Or can I setup the V2Ray client to use my V2Ray server DNS resolution ?
> Thanks !

[image]
You may set your DNS server like this.

iyuangang, 2020-03-06:
If you can not find the custom input quote, change browser or upgrade `ssr plus + `

closed 2020-03-06 by coolsnowwolf