coolsnowwolf/lede #3600: ssr  plus GFW列表模式和全局模式流控问题

djs910, 2020-03-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [   ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今早编译的固件运行 ssr 服务，无论使用GFW列表模式还是全局模式好像是流控出现问题台。
台湾节点选GFW列表模式，手机测速还是显示台湾 PQS-NET地址。手机从启动后第一次显示地址对，再次测速有显示PQS-NET。另外使用google vps 的trojan，全局模式上不了speedtest.cn和youtube，
GFW模式能上。

### (2) 路由器型号和固件版本

A：*86软路由

### (3) 详细日志

A：

djs910, 2020-03-06:
卸载luci-app-ssr-plus，安装之前的luci-app-ssr-plus 158-1 就解决问题了

chchia, 2020-03-06:
我是安装ipt2socks,163-1也可以跑trojan了。

chchia, 2020-03-06:
不好意思，虽然手动添加了ipt2socks但是并没有完全解决问题。

Trojan 在 luci-app-ssr-plus 158-1 之后就不能使用了，只有158-1才是稳定的，之后版本要不完全不能用要不用了一会就不能用。手动添加ipt2socks并不能完美解决。

hxlls, 2020-03-06:
[image]
测试全局模式是没问题的，speedtest.cn应该是去广告规则去掉了，你可以尝试下不保存配置更新一次，试试看有没有问题，因为里面的规则更改太多，不建议保存配置升级

closed 2020-03-06 by coolsnowwolf