coolsnowwolf/lede #3603: SSR+ SCOKS5代理 SSL handshake failed with xxx.xxx.xxx.xxx:443: stream truncated，无法连接

WellsWang, 2020-03-06:
1. 本地局域网使用trojan-cli linux x64版本建立一个本地的代理，端口1080（同一网段）
2. LEDE 配置 SOCKS5 代理连接到trojan-cli proxy，访问时，trojan-cli端报错如下：
[2020-03-06 11:25:05] [INFO] 192.168.1.1:59636 disconnected, 0 bytes received, 517 bytes sent, lasted for 28 seconds
[2020-03-06 11:25:05] [ERROR] 192.168.1.1:59634 SSL handshake failed with xxx.xxx.xxx.xxx:443: stream truncated

3. 同一时间点如果使用Chrome浏览器的SwitchyOmega插件直接连接trojan-cli的Socks5代理可以成功访问，并不报错。

4. 路由器、代理服务器、客户端都在同一网段，当中没有防火墙

5. trojan-cli运行在路由器上（同样也是x64版本）则没有这个报错。

trojan-cli配置如下：

{
    "run_type": "client",
    "local_addr": "0.0.0.0",
    "local_port": 1080,
    "remote_addr": "xxxxxxxxx",
    "remote_port": 443,
    "password": [
        "xxxxxxx"
    ],
    "log_level": 1,
    "ssl": {
        "verify": true,
        "verify_hostname": true,
        "cert": "fullchain.cer",
        "cipher_tls13":"TLS_AES_128_GCM_SHA256:TLS_CHACHA20_POLY1305_SHA256:TLS_AES_256_GCM_SHA384",
"sni": "",
        "alpn": [
            "h2",
            "http/1.1"
        ],
        "reuse_session": true,
        "session_ticket": false,
        "curves": ""
    },
    "tcp": {
        "no_delay": true,
        "keep_alive": true,
        "fast_open": false,
        "fast_open_qlen": 20
    }
}

coolsnowwolf, 2020-03-06:
这个你需要问问 Trojan 那边了，用 SSR 内置的测试通过，我固件自带的 Socks5 也通过

coolsnowwolf, 2020-03-06:
另外注意的内网的 Trojan SOCKS5 要加入到 OP 白名单客户端，否则有可能自环了

chchia, 2020-03-06:
Trojan 在 luci-app-ssr-plus 158-1 之后就不能使用了，只有158-1才是稳定的，之后版本要不完全不能用要不用了一会就不能用。手动添加ipt2socks并不能完美解决。

icybest, 2020-03-06:
换redsocks2后给trojan做udp透明代理似乎是有点问题。

SharerMax, 2020-03-06:
> Trojan 在 luci-app-ssr-plus 158-1 之后就不能使用了，只有 158-1 才是稳定的，之后版本要不完全不能用要不用了一会就不能用。手动添加 ipt2socks 并不能完美解决。

怎么个不能用

hcym, 2020-03-06:
下午编译的正常使用

chchia, 2020-03-06:
> 下午编译的正常使用

刚刚也编译了最新的，暂时用了半小时没问题。

closed 2020-03-06 by coolsnowwolf