coolsnowwolf/lede #3613: 网易云音乐golang版出现错误无法解锁

yinzhefuyun, 2020-03-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天下午更新的解锁网易云音乐golang版出问题，初步感觉是拼接url字符串的时候拼接错误

### (2) 路由器型号和固件版本

A：P&W R619AC 128M

### (3) 详细日志

A：search song:说好不哭 周杰伦
{"Id":"1391274164","Size":43879571,"Br":999000,"Url":"http://su.sycdn.kuwo.cn/ca5bada1ac976bebc23dd6c5eaf991e5/5e624f97/resource/s1/93/41/1323319837.flac","Md5":"","Name":"说好不哭 (with 五月天阿信)","Artist":"周杰伦","MatchScore":0.6057143}
http.Client.Do fail:Get "**http://music.163.comhttp**//su.sycdn.kuwo.cn/ca5bada1ac976bebc23dd6c5eaf991e5/5e624f97/resource/s1/93/41/1323319837.flac": dial tcp: lookup **music.163.comhttp** on 127.0.0.1:53: no such host
Received request CONNECT m8.music.126.net 192.168.1.243:3727
find song info,id: 1391274164
hit cache:

coolsnowwolf, 2020-03-06:
你的路由器 /etc/host 文件有问题，请自查

yinzhefuyun, 2020-03-06:
> 你的路由器 /etc/host 文件有问题，请自查

/etc/下只有hosts文件内容如下：

127.0.0.1 localhost

::1     localhost ip6-localhost ip6-loopback
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

Ryuchen, 2020-03-06:
同样无法解锁

coolsnowwolf, 2020-03-06:
去作者主页提交，不是我写的

closed 2020-03-06 by coolsnowwolf

yinzhefuyun, 2020-03-09:
> 去作者主页提交，不是我写的
早上提了
https://github.com/cnsilvan/UnblockNeteaseMusic/commit/8dd3f9779fcdaa271414a85b80f43f11300e08d3
原作者修复了