coolsnowwolf/lede #3621: 昨晚编译的版本开启SSR和adbyebye，CPU占用率超高

ricksuzade-maker, 2020-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：昨晚编译的版本开启SSR和adbyebye，CPU占用率超高，经过路由器top命令检查后发现是v2ray-plugin一直高占用CPU，前两天的版本没有发现这个问题。

### (2) 路由器型号和固件版本

A：intel i5-7200u软路由，VMware 6.7虚拟机，固件版本2020-3-6日最后一个版本。

### (3) 详细日志

A：[    0.000000] Linux version 4.19.106 (runner@fv-az88) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r0-d8020f6)) #0 SMP Fri Mar 6 13:46:07 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=607C3638-7621-4C9A-A1DD-650CEEE20002 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] [Firmware Warn]: BIOS disabled fast string operation, re-enable failed.
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009f3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bfedffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bfee0000-0x00000000bfefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bfeff000-0x00000000bfefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bff00000-0x00000000bfffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 12/12/2018
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 2712.000 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using sched offset of 13223458955 ns
[    0.000001] tsc: Detected 2712.000 MHz processor
[    0.004049] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.004050] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.004054] last_pfn = 0xc0000 max_arch_pfn = 0x400000000
[    0.004068] MTRR default type: uncachable
[    0.004069] MTRR fixed ranges enabled:
[    0.004069]   00000-9FFFF write-back
[    0.004070]   A0000-BFFFF uncachable
[    0.004070]   C0000-CBFFF write-protect
[    0.004071]   CC000-EFFFF uncachable
[    0.004071]   F0000-FFFFF write-protect
[    0.004072] MTRR variable ranges enabled:
[    0.004073]   0 base 00000000000 mask 7FF00000000 write-back
[    0.004073]   1 base 000C0000000 mask 7FFC0000000 uncachable
[    0.004074]   2 disabled
[    0.004074]   3 disabled
[    0.004074]   4 disabled
[    0.004074]   5 disabled
[    0.004075]   6 disabled
[    0.004075]   7 disabled
[... 1177 lines cut ...]
[   15.966198] pppoe-wan: renamed from ppp0
[   16.010482] pppoe-wan2: renamed from ppp1
[   16.060865] pppoe-vwan1: renamed from ppp2
[   35.219002] sh (7798): drop_caches: 3
[   39.663703] sh (9015): drop_caches: 3
[   42.760416] sh (10516): drop_caches: 3
[   46.059716] sh (11050): drop_caches: 3
[   48.990221] sh (11566): drop_caches: 3
[   51.565233] sh (11984): drop_caches: 3
[   54.392066] sh (13458): drop_caches: 3
[   57.846246] sh (16610): drop_caches: 3
[   60.457452] sh (17034): drop_caches: 3
[   62.871376] sh (17550): drop_caches: 3
[   94.896766] F2FS-fs (sda3): Magic Mismatch, valid(0xf2f52010) - read(0x89e44589)
[   94.896923] F2FS-fs (sda3): Can't find valid F2FS filesystem in 1th superblock
[   94.897213] F2FS-fs (sda3): Magic Mismatch, valid(0xf2f52010) - read(0xe7f202a6)
[   94.897358] F2FS-fs (sda3): Can't find valid F2FS filesystem in 2th superblock
[   94.897502] F2FS-fs (sda3): Magic Mismatch, valid(0xf2f52010) - read(0x89e44589)
[   94.897644] F2FS-fs (sda3): Can't find valid F2FS filesystem in 1th superblock
[   94.897784] F2FS-fs (sda3): Magic Mismatch, valid(0xf2f52010) - read(0xe7f202a6)
[   94.897926] F2FS-fs (sda3): Can't find valid F2FS filesystem in 2th superblock
[   94.898193] [EXFAT] trying to mount...
[   94.898531] ntfs: (device sda3): parse_options(): Unrecognized mount option discard.
[ 1024.383254] nf_ct_snmp: dropping packet: parser failed
[ 1024.383254]  IN= OUT= SRC=118.114.170.11 DST=113.117.7.206 LEN=48 TOS=0x00 PREC=0x00 TTL=127 ID=17139 PROTO=UDP SPT=63960 DPT=161 LEN=28 MARK=0x300

blankhang, 2020-03-07:
这几天的版本 好像 V2RAY 都会自动断 虽然显示正在运行中 但服务貌似已挂

georip, 2020-03-07:
ADBYBY,先关闭，再开启就可以了。重启路由器后adbyby继续打满100%。。。。这问题出现好久了。我的路由器是竞斗云。。。……

blankhang, 2020-03-07:
刚刚拉了最新的代码编译 貌似老大已经把这个插件移除了...

coolsnowwolf, 2020-03-07:
删除 APP Filter 就好了

ricksuzade-maker, 2020-03-07:
哇塞，L大亲自回复，我是新手，敢问APP Filter在哪里删？ 我找了一遍也没找到，网上也没搜到答案，感谢！

closed 2020-03-08 by coolsnowwolf