coolsnowwolf/lede #3623: v2ray模式下无法使用mkcp协议进行科学上网

hhmaomao, 2020-03-07:
反馈bug/问题模板，提建议请删除

1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

(1) 具体问题

A：编译固件后，发现kcptun状态显示未运行，编译固件时已勾选kcptun。通过更改vps的v2ray传输协议，更改为mkcp后，写入路由器，发现无法使用mkcp协议进行科学上网，协议改为kcp或ws+tls可以正常科学上网，希望求解

(2) 路由器型号和固件版本

A：X86路由器

### (3) 详细日志

A：

hhmaomao, 2020-03-07:
**(3) 详细日志**
Sat Mar  7 13:46:28 2020 daemon.err uhttpd[2739]: cat: write error: Broken pipe
Sat Mar  7 13:48:53 2020 daemon.warn pdnsd[5188]: Caught signal 15. Exiting.
Sat Mar  7 13:48:53 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar  7 13:48:53 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Mar  7 13:48:56 2020 cron.info crond[18725]: crond (busybox 1.30.0) started, log level 8
Sat Mar  7 13:48:58 2020 daemon.info pdnsd[18947]: pdnsd-1.2.9b-par starting.
Sat Mar  7 13:48:59 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar  7 13:48:59 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Mar  7 13:49:35 2020 daemon.warn pdnsd[18947]: Caught signal 15. Exiting.
Sat Mar  7 13:49:35 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar  7 13:49:35 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Mar  7 13:49:38 2020 cron.info crond[19799]: crond (busybox 1.30.0) started, log level 8
Sat Mar  7 13:49:40 2020 daemon.info pdnsd[20030]: pdnsd-1.2.9b-par starting.
Sat Mar  7 13:49:41 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar  7 13:49:41 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

coolsnowwolf, 2020-03-07:
你对端 v2ray 也必须是 mkcp

hhmaomao, 2020-03-07:
我已经修改了vps和固件上的协议为mkcp，但是依然无法使用很是奇怪，刚开始以为是系统问题，重装了依然没有解决，最后换了几个服务商，瓦工，cloudcone，hostdare的vps都这样，全部都重装系统，然后安装v2使用kcp协议，一样无法解决问题，甚是奇怪

hyird, 2020-03-07:
自己电脑测试过没有

hhmaomao, 2020-03-07:
自己电脑手机测试均正常，所以才发的求助

closed 2020-03-09 by coolsnowwolf

blizzardster, 2020-10-04:
> 反馈bug/问题模板，提建议请删除
> 
> 1.关于你要提交的问题
> 
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> ## 2. 详细叙述
> (1) 具体问题
> 
> A：编译固件后，发现kcptun状态显示未运行，编译固件时已勾选kcptun。通过更改vps的v2ray传输协议，更改为mkcp后，写入路由器，发现无法使用mkcp协议进行科学上网，协议改为kcp或ws+tls可以正常科学上网，希望求解
> 
> (2) 路由器型号和固件版本
> 
> A：X86路由器
> 
> ### (3) 详细日志
> A：

是不是mkcp的kcp seed密码没填，在固件中我找不到哪里填写