coolsnowwolf/lede #3625: SSR PLUS+ 游戏增强模式UDP失效

roy63258015, 2020-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ssr plus+ 游戏增强模式 UDP 失效 switch NAT F

### (2) 路由器型号和固件版本

A：X86

固件版本 | OpenWrt R20.3.5 / LuCI Master (git-20.062.50551-a6189c5)
-- | --
内核版本 | 4.19.106

### (3) 详细日志

A：
Sat Mar  7 13:40:24 2020 daemon.err uhttpd[2476]: uci: Parse error (invalid command) at line 1, byte 0
Sat Mar  7 13:40:24 2020 daemon.err uhttpd[2476]: uci: Parse error (invalid command) at line 1, byte 0
Sat Mar  7 13:40:24 2020 daemon.err uhttpd[2476]: uci: Parse error (invalid command) at line 1, byte 0
Sat Mar  7 13:40:24 2020 daemon.err uhttpd[2476]: uci: Entry not found
Sat Mar  7 13:40:24 2020 daemon.err uhttpd[2476]: uci: Entry not found
Sat Mar  7 13:40:24 2020 daemon.err uhttpd[2476]: uci: Entry not found
Sat Mar  7 13:40:56 2020 cron.info crond[24130]: crond (busybox 1.30.0) started, log level 8
Sat Mar  7 13:41:16 2020 cron.info crond[24415]: crond (busybox 1.30.0) started, log level 8
Sat Mar  7 13:42:48 2020 cron.info crond[24853]: crond (busybox 1.30.0) started, log level 8
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25060]: initializing ciphers... aes-256-gcm
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25060]: listening at :::1234
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25060]: tcp port reuse enabled
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25060]: running from root user
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25062]: initializing ciphers... aes-256-gcm
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25062]: listening at :::1234
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25062]: tcp port reuse enabled
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25062]: running from root user
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25064]: initializing ciphers... aes-256-gcm
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25064]: listening at :::1234
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25064]: tcp port reuse enabled
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25064]: running from root user
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25066]: initializing ciphers... aes-256-gcm
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25066]: listening at :::1234
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25066]: tcp port reuse enabled
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25066]: running from root user
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25093]: initializing ciphers... aes-256-gcm
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25093]: listening at :::1234
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25093]: UDP relay enabled
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25093]: udp port reuse enabled
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25093]: TCP relay disabled
Sat Mar  7 13:42:49 2020 daemon.info /usr/bin/ss-redir[25093]: running from root user
Sat Mar  7 13:42:49 2020 daemon.info dnsmasq[23656]: exiting on receipt of SIGTERM
Sat Mar  7 13:42:49 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Mar  7 13:42:49 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: started, version 2.80 cachesize 3000
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: DNS service limited to local subnets
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify dumpfile
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq-dhcp[25299]: DHCP, IP range 192.168.0.100 -- 192.168.0.249, lease time 12h
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using local addresses only for domain test
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using local addresses only for domain onion
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using local addresses only for domain localhost
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using local addresses only for domain local
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using local addresses only for domain invalid
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using local addresses only for domain bind
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using nameserver 127.0.0.1#5335 for domain edgekey.net
Sat Mar  7 13:42:53 2020 daemon.info dnsmasq[25299]: using nameserver 127.0.0.1#5335 for domain edgesuite.net
[... 598 lines cut ...]
Sat Mar  7 13:51:31 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/odhcpd - 6 addresses
Sat Mar  7 13:51:31 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/dhcp.cfg01411c - 33 addresses
Sat Mar  7 13:51:31 2020 daemon.info dnsmasq-dhcp[28133]: read /etc/ethers - 0 addresses
Sat Mar  7 13:52:00 2020 daemon.err uhttpd[2476]: sed: write error
Sat Mar  7 13:53:10 2020 daemon.info dnsmasq[28133]: read /etc/hosts - 4 addresses
Sat Mar  7 13:53:10 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/odhcpd - 6 addresses
Sat Mar  7 13:53:10 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/dhcp.cfg01411c - 33 addresses
Sat Mar  7 13:53:10 2020 daemon.info dnsmasq-dhcp[28133]: read /etc/ethers - 0 addresses
Sat Mar  7 13:53:32 2020 daemon.info dnsmasq[28133]: read /etc/hosts - 4 addresses
Sat Mar  7 13:53:32 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/odhcpd - 6 addresses
Sat Mar  7 13:53:32 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/dhcp.cfg01411c - 33 addresses
Sat Mar  7 13:53:32 2020 daemon.info dnsmasq-dhcp[28133]: read /etc/ethers - 0 addresses
Sat Mar  7 13:54:35 2020 daemon.info dnsmasq[28133]: read /etc/hosts - 4 addresses
Sat Mar  7 13:54:35 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/odhcpd - 6 addresses
Sat Mar  7 13:54:35 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/dhcp.cfg01411c - 33 addresses
Sat Mar  7 13:54:35 2020 daemon.info dnsmasq-dhcp[28133]: read /etc/ethers - 0 addresses
Sat Mar  7 13:55:51 2020 daemon.info dnsmasq[28133]: read /etc/hosts - 4 addresses
Sat Mar  7 13:55:51 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/odhcpd - 6 addresses
Sat Mar  7 13:55:51 2020 daemon.info dnsmasq[28133]: read /tmp/hosts/dhcp.cfg01411c - 33 addresses
Sat Mar  7 13:55:51 2020 daemon.info dnsmasq-dhcp[28133]: read /etc/ethers - 0 addresses
Sat Mar  7 13:57:27 2020 daemon.info dnsmasq-dhcp[28133]: DHCPDISCOVER(br-lan) d4:f0:57:f2:a7:3d
Sat Mar  7 13:57:27 2020 daemon.info dnsmasq-dhcp[28133]: DHCPOFFER(br-lan) 192.168.0.31 d4:f0:57:f2:a7:3d
Sat Mar  7 13:57:27 2020 daemon.info dnsmasq-dhcp[28133]: DHCPREQUEST(br-lan) 192.168.0.31 d4:f0:57:f2:a7:3d
Sat Mar  7 13:57:27 2020 daemon.info dnsmasq-dhcp[28133]: DHCPACK(br-lan) 192.168.0.31 d4:f0:57:f2:a7:3d switch
Sat Mar  7 13:58:02 2020 daemon.err /usr/bin/ss-redir[27914]: server recv: Connection reset by peer

coolsnowwolf, 2020-03-07:
默认编译

roy63258015, 2020-03-07:
好的 默认编译测试下 看看能不能解决

roy63258015, 2020-03-07:
> 默认编译

UDP还是不行
系统日志
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.445412] PCI host bridge to bus 0000:00
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.449629] pci_bus 0000:00: root bus resource [io  0x0070-0x0077]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.454817] pci_bus 0000:00: root bus resource [io  0x0000-0x006f window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.461724] pci_bus 0000:00: root bus resource [io  0x0078-0x0cf7 window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.474817] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.481721] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.484816] pci_bus 0000:00: root bus resource [mem 0x000c0000-0x000dffff window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.492484] pci_bus 0000:00: root bus resource [mem 0x000e0000-0x000fffff window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.504816] pci_bus 0000:00: root bus resource [mem 0xd0c00000-0xd0ffffff window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.512485] pci_bus 0000:00: root bus resource [mem 0x80000000-0xd0a15fff window]
Sat Mar  7 14:40:45 2020 kern.info kernel: [    1.514816] pci_bus 0000:00: root bus resource [bus 00-ff]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.520433] pci 0000:00:00.0: [8086:0f00] type 00 class 0x060000
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.520658] pci 0000:00:02.0: [8086:0f31] type 00 class 0x030000
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.520674] pci 0000:00:02.0: reg 0x10: [mem 0xd0000000-0xd03fffff]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.520685] pci 0000:00:02.0: reg 0x18: [mem 0xc0000000-0xcfffffff pref]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.520696] pci 0000:00:02.0: reg 0x20: [io  0xf080-0xf087]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: [8086:0f23] type 00 class 0x010601
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: reg 0x10: [io  0xf070-0xf077]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: reg 0x14: [io  0xf060-0xf063]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: reg 0x18: [io  0xf050-0xf057]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: reg 0x1c: [io  0xf040-0xf043]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: reg 0x20: [io  0xf020-0xf03f]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: reg 0x24: [mem 0xd0a15000-0xd0a157ff]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.524811] pci 0000:00:13.0: PME# supported from D3hot
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528420] pci 0000:00:14.0: [8086:0f35] type 00 class 0x0c0330
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528420] pci 0000:00:14.0: reg 0x10: [mem 0xd0a00000-0xd0a0ffff 64bit]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528420] pci 0000:00:14.0: PME# supported from D3hot D3cold
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528669] pci 0000:00:1a.0: [8086:0f18] type 00 class 0x108000
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528669] pci 0000:00:1a.0: reg 0x10: [mem 0xd0500000-0xd05fffff]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528669] pci 0000:00:1a.0: reg 0x14: [mem 0xd0400000-0xd04fffff]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528669] pci 0000:00:1a.0: PME# supported from D0 D3hot
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528997] pci 0000:00:1b.0: [8086:0f04] type 00 class 0x040300
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528997] pci 0000:00:1b.0: reg 0x10: [mem 0xd0a10000-0xd0a13fff 64bit]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.528997] pci 0000:00:1b.0: PME# supported from D0 D3hot D3cold
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.529288] pci 0000:00:1c.0: [8086:0f48] type 01 class 0x060400
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.529288] pci 0000:00:1c.0: PME# supported from D0 D3hot D3cold
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.529659] pci 0000:00:1c.1: [8086:0f4a] type 01 class 0x060400
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.529659] pci 0000:00:1c.1: PME# supported from D0 D3hot D3cold
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.529950] pci 0000:00:1c.2: [8086:0f4c] type 01 class 0x060400
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.529950] pci 0000:00:1c.2: PME# supported from D0 D3hot D3cold
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.530211] pci 0000:00:1c.3: [8086:0f4e] type 01 class 0x060400
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.530211] pci 0000:00:1c.3: PME# supported from D0 D3hot D3cold
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.530470] pci 0000:00:1f.0: [8086:0f1c] type 00 class 0x060100
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.530727] pci 0000:00:1f.3: [8086:0f12] type 00 class 0x0c0500
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.530727] pci 0000:00:1f.3: reg 0x10: [mem 0xd0a14000-0xd0a1401f]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.530727] pci 0000:00:1f.3: reg 0x20: [io  0xf000-0xf01f]
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.531242] pci 0000:01:00.0: [8086:150c] type 00 class 0x020000
Sat Mar  7 14:40:45 2020 kern.debug kernel: [    1.531242] pci 0000:01:00.0: reg 0x10: [mem 0xd0900000-0xd091ffff]
[... cut ...]
Sat Mar  7 14:41:36 2020 cron.info crond[11901]: crond (busybox 1.30.0) started, log level 8
Sat Mar  7 14:41:36 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sat Mar  7 14:41:36 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Sat Mar  7 14:41:36 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sat Mar  7 14:41:36 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Sat Mar  7 14:41:36 2020 daemon.notice netifd: Interface 'wan_6' is now up

Powered by LuCI Master (git-20.062.50551-a6189c5) / OpenWrt R20.3.5

roy63258015, 2020-03-07:
163-1 没问题 NAT正常

BLUEHAT25, 2020-03-07:
这个log是真的长。。。。

gpeditmsc1986, 2020-03-07:
你是怎么编译的，有十多个错误日志

coolsnowwolf, 2020-03-08:
和 IPv6 冲突了，去掉 IPv6 功能即可恢复，git pull 即可

roy63258015, 2020-03-08:
最新的164-2的udp还是有问题，不过switch 从nat f提高到nat D

closed 2020-03-08 by coolsnowwolf

hongwen000, 2022-05-04:
wsdd2占5335端口？