coolsnowwolf/lede #3626: 解锁网易云灰色歌曲 开启后云盘上传失败

blankhang, 2020-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
解锁网易云灰色歌曲 开启后云盘上传音乐会失败  
如果关掉则能正常上传

### (2) 路由器型号和固件版本

* 软路由 Intel(R) Celeron(R) CPU N3150 
* 固件版本 OpenWrt R20.3.5 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志
#### 系统日志
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.401685] pci 0000:00:18.6: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.403578] pci 0000:00:18.6: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.405832] pci 0000:00:18.5: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.407739] pci 0000:00:18.5: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.409930] pci 0000:00:18.4: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.411832] pci 0000:00:18.4: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.413845] pci 0000:00:18.3: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.415774] pci 0000:00:18.3: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.418038] pci 0000:00:18.2: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.419934] pci 0000:00:18.2: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.422288] pci 0000:00:17.7: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.424373] pci 0000:00:17.7: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.426750] pci 0000:00:17.6: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.428880] pci 0000:00:17.6: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.430876] pci 0000:00:17.5: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.433165] pci 0000:00:17.5: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.435178] pci 0000:00:17.4: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.437309] pci 0000:00:17.4: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.439316] pci 0000:00:17.3: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.441224] pci 0000:00:17.3: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.443210] pci 0000:00:16.7: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.445510] pci 0000:00:16.7: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.447510] pci 0000:00:16.6: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.449785] pci 0000:00:16.6: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.452227] pci 0000:00:16.5: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.454165] pci 0000:00:16.5: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.456846] pci 0000:00:16.4: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.458793] pci 0000:00:16.4: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.460803] pci 0000:00:16.3: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.462704] pci 0000:00:16.3: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.464756] pci 0000:00:15.7: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.466897] pci 0000:00:15.7: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.469140] pci 0000:00:15.6: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.471069] pci 0000:00:15.6: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.473067] pci 0000:00:15.5: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.474980] pci 0000:00:15.5: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.477183] pci 0000:00:15.4: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.479362] pci 0000:00:15.4: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.481389] pci 0000:00:15.3: BAR 7: no space for [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.483363] pci 0000:00:15.3: BAR 7: failed to assign [io  size 0x1000]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.485416] pci 0000:00:01.0: PCI bridge to [bus 01]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.487642] pci 0000:02:01.0: BAR 6: assigned [mem 0xfd500000-0xfd50ffff pref]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.489867] pci 0000:02:05.0: BAR 6: assigned [mem 0xfd510000-0xfd51ffff pref]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.492107] pci 0000:00:11.0: PCI bridge to [bus 02]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.493665] pci 0000:00:11.0:   bridge window [io  0x2000-0x3fff]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.495682] pci 0000:00:11.0:   bridge window [mem 0xfd500000-0xfdffffff]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.497755] pci 0000:00:11.0:   bridge window [mem 0xe7b00000-0xe7ffffff 64bit pref]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.500168] pci 0000:00:15.0: PCI bridge to [bus 03]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.501699] pci 0000:00:15.0:   bridge window [io  0x4000-0x4fff]
Sat Mar  7 13:21:49 2020 kern.info kernel: [    1.504078] pci 0000:00:15.0:   bridge window [mem 0xfd400000-0xfd4fffff]
[... 1786 lines cut ...]
[   10.965229] usbcore: registered new interface driver snd-usb-audio
[   11.000147] usbcore: registered new interface driver rt2800usb
[   11.007079] usbcore: registered new interface driver ath9k_htc
[   11.013439] kmodloader: done loading kernel modules from /etc/modules.d/*
[   28.898661] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[   28.898661] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   29.956971] 8021q: adding VLAN 0 to HW filter on device eth0
[   29.959367] br-lan: port 1(eth0) entered blocking state
[   29.961000] br-lan: port 1(eth0) entered disabled state
[   29.965314] device eth0 entered promiscuous mode
[   29.969855] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   29.975275] br-lan: port 1(eth0) entered blocking state
[   29.977176] br-lan: port 1(eth0) entered forwarding state
[   29.979008] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   29.994709] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   29.996635] 8021q: adding VLAN 0 to HW filter on device eth1
[   30.007722] e1000: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   30.011878] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   30.977138] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   34.657727] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[   38.573679] br-lan: port 1(eth0) entered disabled state
[   38.585909] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   38.589484] br-lan: port 1(eth0) entered blocking state
[   38.591197] br-lan: port 1(eth0) entered forwarding state
[   38.726188] e1000: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None

hyird, 2020-03-07:
都劫持了 ，正常  ， 手机app也没法登陆

coolsnowwolf, 2020-03-07:
我感觉这个功能白做了。云盘走 HTTPS，不劫持 HTTPS 即可

[image]

coolsnowwolf, 2020-03-07:
安卓手机也无需代理 HTTPS

blankhang, 2020-03-07:
感谢大佬回复 .我试过了 不代理HTTPS 也不行?
不代理HTTP 正常
但不代理HTTP 无法解锁了

> 我感觉这个功能白做了。云盘走 HTTPS，不劫持 HTTPS 即可
> 
> [image]

coolsnowwolf, 2020-03-08:
Windows 加装证书

blankhang, 2020-03-08:
[image]
请问大佬跟网易云音乐版本有关吗 现在是2.7.1 198242

blankhang, 2020-03-08:
我也有请教过原开发大佬
这是开发大佬这边的回复 请参考
查了下应该是 node 和 go 的版本都集成了
我自己的云盘正常
不过我只使用代理，不是 ipset 劫持
这个和证书没关系

closed 2020-03-08 by coolsnowwolf

blankhang, 2020-03-08:
? 问题关闭就解决了 ?