coolsnowwolf/lede #3631: 小白自己编译固件，最后一步总是报错，请大佬指点意思

CarrolChen, 2020-03-07:
Make V=99报错反馈如下：
src/gpsdclient.c: In function 'nmeaInfoFromGpsd':
src/gpsdclient.c:291:101: warning: format '%f' expects argument of type 'double', but argument 12 has type 'timespec_t {aka struct timespec}' [-Wformat=]
         syslog(LOG_INFO, "Using %s device%s%s on %s in %s mode at %u baud (%u%c%u), refresh time %.3f (min. %.3f)", //
                                                                                                  ~~~^
src/gpsdclient.c:301:13:
             dev->cycle, //
             ~~~~~~~~~~                                                                               
src/gpsdclient.c:291:112: warning: format '%f' expects argument of type 'double', but argument 13 has type 'timespec_t {aka struct timespec}' [-Wformat=]
         syslog(LOG_INFO, "Using %s device%s%s on %s in %s mode at %u baud (%u%c%u), refresh time %.3f (min. %.3f)", //
                                                                                                             ~~~^
src/gpsdclient.c:302:13:
             dev->mincycle);
             ~~~~~~~~~~~~~                                                                                       
In file included from ../../lib/pud/nmealib/include/nmealib/info.h:21:0,
                 from src/gpsdclient.h:51,
                 from src/gpsdclient.c:46:
../../lib/pud/nmealib/include/nmealib/util.h:48:21: error: invalid operands to binary != (have 'timespec_t {aka struct timespec}' and 'timespec_t {aka struct timespec}')
 #define isNaN(x) (x != x)
                     ^
src/gpsdclient.c:370:8: note: in expansion of macro 'isNaN'
   if (!isNaN(gpsdata->fix.time)) {
        ^~~~~
src/gpsdclient.c:372:33: error: incompatible type for argument 1 of 'fabs'
     double fraction = modf(fabs(gpsdata->fix.time), &seconds);
                                 ^~~~~~~
In file included from ../../lib/pud/wireformat/include/OlsrdPudWireFormat/wireFormat.h:57:0,
                 from src/configuration.h:59,
                 from src/gpsdclient.c:48:
/home/carrol/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include/math.h:203:13: note: expected 'double' but argument is of type 'timespec_t {aka struct timespec}'
 double      fabs(double);
             ^~~~
src/gpsdclient.c:390:7: error: wrong type argument to unary exclamation mark
   if (!gpsdata->online) {
       ^
src/gpsdclient.c:457:29: error: 'struct gps_data_t' has no member named 'separation'; did you mean 'version'?
     info->height = gpsdata->separation;
                             ^~~~~~~~~~
                             version
src/gpsdclient.c: In function 'readFromGpsd':
src/gpsdclient.c:581:19: error: too few arguments to function 'gps_read'
     gpsReadCode = gps_read(gpsdata);
                   ^~~~~~~~
In file included from src/gpsdclient.h:49:0,
                 from src/gpsdclient.c:46:
/home/carrol/lede/staging_dir/target-mipsel_24kc_musl/usr/include/gps.h:2307:12: note: declared here
 extern int gps_read(struct gps_data_t *, char *message, int message_len);
            ^~~~~~~~
../../Makefile.inc:351: recipe for target 'src/gpsdclient.o' failed
make[5]: *** [src/gpsdclient.o] Error 1
make[5]: Leaving directory '/home/carrol/lede/build_dir/target-mipsel_24kc_musl/olsrd-0.9.7/lib/pud'
Makefile:241: recipe for target 'libs' failed
make[4]: *** [libs] Error 2
make[4]: Leaving directory '/home/carrol/lede/build_dir/target-mipsel_24kc_musl/olsrd-0.9.7'
Makefile:293: recipe for target '/home/carrol/lede/build_dir/target-mipsel_24kc_musl/olsrd-0.9.7/.built' failed
make[3]: *** [/home/carrol/lede/build_dir/target-mipsel_24kc_musl/olsrd-0.9.7/.built] Error 2
make[3]: Leaving directory '/home/carrol/lede/feeds/routing/olsrd'
time: package/feeds/routing/olsrd/compile#16.78#4.13#21.58
package/Makefile:111: recipe for target 'package/feeds/routing/olsrd/compile' failed
make[2]: *** [package/feeds/routing/olsrd/compile] Error 2
make[2]: Leaving directory '/home/carrol/lede'
package/Makefile:107: recipe for target '/home/carrol/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/carrol/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/carrol/lede'
/home/carrol/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

已经试了好几次，都在最后一步报错，网上都没找到解决办法，实在没招了，请教一下大佬，望大佬不吝赐教，拜托了！

guochan2019, 2020-03-07:
感觉像是科学上网的问题，组件拉不下来，建议make menuconfig后，最好执行一下make download，然后用WinSCP或其他工具登录到虚拟机看dl目录下有无1k字节的文件，如果有说明当前部分文件无法拉回，换个时间再试，或从其他渠道取回。仅供参考

CarrolChen, 2020-03-07:
> 感觉像是科学上网的问题，组件拉不下来，建议make menuconfig后，最好执行一下make download，然后用WinSCP或其他工具登录到虚拟机看dl目录下有无1k字节的文件，如果有说明当前部分文件无法拉回，换个时间再试，或从其他渠道取回。

make download我在make V=99编译之前已经执行过了，全部通过，没有问题，所以自己试了好几次一直不行才来求教的，网上也没有对这个问题出现的case作说明，试了好几天了，希望哪个大佬看到，求教一下

dolphinpaopao, 2020-03-07:
最近几次测试用海外VPS编译，没有失败过。阿里云HK 8核8G 40GB SSD，大约￥2.5/h。国内网络环境众所周知，编译的时候在下载阶段容易出现很多莫名其妙的问题。你也可以尝试一下，这样可以排除很多问题。编译遇到报错，没有头绪的时候，首先删除原代码所有文件并全新拉取最新代码默认编译一次，假如不通过，贴出报错，这样方便定位错误原因。

AmadeusGhost, 2020-03-07:
别选跟olsr相关的包

CarrolChen, 2020-03-07:
> 最近几次测试用海外VPS编译，没有失败过。阿里云HK 8核8G 40GB SSD，大约￥2.5/h。国内网络环境众所周知，编译的时候在下载阶段容易出现很多莫名其妙的问题。你也可以尝试一下，这样可以排除很多问题。编译遇到报错，没有头绪的时候，首先删除原代码所有文件并全新拉取最新代码默认编译一次，假如不通过，贴出报错，这样方便定位错误原因。

已经摸索差不多一周时间了，下班有时间就去看一些case，没试过用海外，时间太长。其次你说的删除源码，多来几次，我真的很负责的讲，我自己都记不清试了多少次，就一直是这个问题。不但这样，虚拟系统我都换了好几次。真的是没办法了，才在这里求助，知道这里大佬多，把问题暴露出来，让更多人看到，帮助小白解决问题的几率就越大。而且很多人，我看他们视频编译中也约到这样的问题，但就是没人正式的提出这个问题，也不知道他们怎么解决的。

CarrolChen, 2020-03-07:
> 别选跟olsr相关的包

关键是怎么去定位这个包是否选择了呢？分析问题的最终目的是解决这个问题，中间还是需要一个可行的做的过程，能否共同具体跑一下？

dolphinpaopao, 2020-03-07:
> 已经摸索差不多一周时间了，下班有时间就去看一些case，没试过用海外，时间太长。其次你说的删除源码，多来几次，我真的很负责的讲，我自己都记不清试了多少次，就一直是这个问题。不但这样，虚拟系统我都换了好几次。真的是没办法了，才在这里求助，知道这里大佬多，把问题暴露出来，让更多人看到，帮助小白解决问题的几率就越大。而且很多人，我看他们视频编译中也约到这样的问题，但就是没人正式的提出这个问题，也不知道他 [...]

虚拟机的系统版本编译环境是什么？我用VB+18.04 x64 server，分配4核心+8GB内存，全程出国，编译-j1一直都正常。避免使用桌面版本。

CarrolChen, 2020-03-07:
> > > 阿里云HK 8核8G 40GB SSD，大约￥ 2.5 / h。国内网络环境状况，编译的时候在下载阶段容易出现很多莫名其妙的问题。你也可以尝试一下，这样可以排除很多问题。编译遇到报错，没有头绪的时候，首先删除原代码所有文件并全新拉取最新代码替换编译一次，假如不通过，贴出报错，这样方便定位错误原因。
> > 
> > 
> > 已经摸索差不多差不多一周时间了，下班有时间就去看一些情况，没试过用海外，时间太长。其次你说的删除原始码，多来一次，我真的很负责的讲，我自己都记不清试了多少次，就一直一直是这个问题。不但这样，虚拟系统我都换了好几次。真的是没办法了，才在这里求助，知道这里大佬多，把问题暴露出来，让更而且很多人，我看他们视频编译中也约到这样的问题，但就是没人正式的提出这个问题，也不知道他们怎么解决的。
> 
> 我用VB + 18.04 x64服​​务器，分配4核心+ 8GB内存，全程出国，编译-j1一直都正常。避免使用桌面版本。

非常感谢你提供的方法，长时间且多次跑在海外的VPS，可能对于我来说，真的不太合适。我再去其他论坛反馈求助看看，到底啥问题！再次感谢~

martin0607, 2020-03-07:
试试看用Gitpod编译

CarrolChen, 2020-03-07:
> 试试看用Gitpod编译

请问具体如何去操作，之前没用过，刚进入openwrt这个路由系统领域，请赐教~

hyird, 2020-03-07:
olsrd 别选 官方的锅 他升级了  估计是不兼容大雕的 
这个也不知道有什么用  自己去掉就好了

CarrolChen, 2020-03-07:
> olsrd 别选 官方的锅 他升级了 估计是不兼容大雕的
> 这个也不知道有什么用 自己去掉就好了

感谢帮助，已经定位到该包，被我剔除，现在正在恢复环境重新编译，就是第一次你知道，跑的会很慢，还在跑。出结果会把问题总结及方法贴给更多小伙伴，人人为我，我为人人！

closed 2020-03-08 by coolsnowwolf

phorcys, 2020-04-05:
https://github.com/OLSR/olsrd/pull/79
这个pull修复了olsrd gps3.20 api变动