coolsnowwolf/lede #3632: 请教lean大nano Pi R1s在勾选了allwinner a64 h5以后luci里面找不到adbyby是停止支持H5了吗

norikohxb, 2020-03-07:
nano Pi R1s在勾选了allwinner a64 h5以后luci里面找不到adbyby是停止支持H5了吗

norikohxb, 2020-03-07:
如果设备类型不选择allwiner h5的类型，那么adbyby就会出来，如果勾选了allwinner h5的设备，那么adbyby plus这个我最喜欢的插件就找不到了

norikohxb, 2020-03-07:
如果是的话能不能恢复对于nano Pi R1S H5的支持？
因为你的adbyby基本上是我必不可少的东东啊

iyuangang, 2020-03-07:
可以把3月4号以前版本的 `adbyby`  `luci-app-adbyby-plus` 包替换现有包。

norikohxb, 2020-03-07:
因为我是lean的op的更新狂，一天git pull至少3次，都是用最新的lean的系统。虽然也吃过bug的亏，但是追新是乐此不疲。最近2天的git pull我甚至看到adbyby也是一直在更新，欣喜若狂。请问如何找到之前旧的luci-app-adbyby-plus？我硬盘都给lean大最新的覆盖了。
然后lean的新的adbyby是否还可以支持r1s H5，因为这个adbyby是命根子，很好用。

xynu617, 2020-03-07:
> 因为我是lean的op的更新狂，一天git pull至少3次，都是用最新的lean的系统。虽然也吃过bug的亏，但是追新是乐此不疲。最近2天的git pull我甚至看到adbyby也是一直在更新，欣喜若狂。请问如何找到之前旧的luci-app-adbyby-plus？我硬盘都给lean大最新的覆盖了。
> 然后lean的新的adbyby是否还可以支持r1s H5，因为这个adbyby是命根子，很好用。

看我发的那个PR，自己修改下试试看会不会回来

norikohxb, 2020-03-07:
> > 因为我是lean的op的更新狂，一天git pull至少3次，都是用最新的lean的系统。虽然也吃过bug的亏，但是追新是乐此不疲。最近2天的git pull我甚至看到adbyby也是一直在更新，欣喜若狂。请问如何找到之前旧的luci-app-adbyby-plus？我硬盘都给lean大最新的覆盖了。
> > 然后lean的新的adbyby是否还可以支持r1s H5，因为这个adbyby是命根子，很好用。
> 
> 看我发的那个PR，自己修改下试试看会不会回来

请教一下PR是什么的缩写？从哪里可以看到您的PR？

norikohxb, 2020-03-07:
> > 因为我是lean的op的更新狂，一天git pull至少3次，都是用最新的lean的系统。虽然也吃过bug的亏，但是追新是乐此不疲。最近2天的git pull我甚至看到adbyby也是一直在更新，欣喜若狂。请问如何找到之前旧的luci-app-adbyby-plus？我硬盘都给lean大最新的覆盖了。
> > 然后lean的新的adbyby是否还可以支持r1s H5，因为这个adbyby是命根子，很好用。
> 
> 看我发的那个PR，自己修改下试试看会不会回来

谢谢你，我知道pr是pull request了，我现在看看

norikohxb, 2020-03-07:
> > 因为我是lean的op的更新狂，一天git pull至少3次，都是用最新的lean的系统。虽然也吃过bug的亏，但是追新是乐此不疲。最近2天的git pull我甚至看到adbyby也是一直在更新，欣喜若狂。请问如何找到之前旧的luci-app-adbyby-plus？我硬盘都给lean大最新的覆盖了。
> > 然后lean的新的adbyby是否还可以支持r1s H5，因为这个adbyby是命根子，很好用。
> 
> 看我发的那个PR，自己修改下试试看会不会回来

你的pull reuqest都是空的啊
https://github.com/xynu617/ONMP/pulls
https://github.com/xynu617/luci/pulls
https://github.com/xynu617/OpenWrt-CI/pulls
都找了没有

norikohxb, 2020-03-07:
> > 因为我是lean的op的更新狂，一天git pull至少3次，都是用最新的lean的系统。虽然也吃过bug的亏，但是追新是乐此不疲。最近2天的git pull我甚至看到adbyby也是一直在更新，欣喜若狂。请问如何找到之前旧的luci-app-adbyby-plus？我硬盘都给lean大最新的覆盖了。
> > 然后lean的新的adbyby是否还可以支持r1s H5，因为这个adbyby是命根子，很好用。
> 
> 看我发的那个PR，自己修改下试试看会不会回来

https://github.com/coolsnowwolf/lede/pull/3629
找到这个帖子了，
按照这个修改就可以了对吗？

iyuangang, 2020-03-07:
刚提交了个PR，我测试过能修复。等更新吧

xynu617, 2020-03-07:
> > > 因为我是lean的op的更新狂，一天git pull至少3次，都是用最新的lean的系统。虽然也吃过bug的亏，但是追新是乐此不疲。最近2天的git pull我甚至看到adbyby也是一直在更新，欣喜若狂。请问如何找到之前旧的luci-app-adbyby-plus？我硬盘都给lean大最新的覆盖了。
> > > 然后lean的新的adbyby是否还可以支持r1s H5，因为这个adbyby是命根子，很好用。
> > 
> > 
> > 看我发的那个PR，自己修改下试试看会不会回来
> 
> #3629
> 找到这个帖子了，
> 按照这个修改就可以了对吗？

你自己试一下，我不确定···我修改了之后32位X86的adbyby重新出现了，上面那一行和下面几个是对应的，有你那个架构的话，加上应该就好了。     改自己的文件，不要发PR

xynu617, 2020-03-07:
> 刚提交了个PR，我测试过能修复。等更新吧

你这是把L大的代码又改回去了啊

norikohxb, 2020-03-07:
> 刚提交了个PR，我测试过能修复。等更新吧

怎么修复你能教我一下吗，修改那个makefile？吗 请赐教啊。

iyuangang, 2020-03-07:
> > 刚提交了个PR，我测试过能修复。等更新吧
> 
> 怎么修复你能教我一下吗，修改那个makefile？吗 请赐教啊。

自己本地编译的话，修改 package/lean/adbyby/Makefile 你看看两天前的版本，改成那个代码就好了

norikohxb, 2020-03-07:
我把dpend前面加上井号就出来了，但是还没有编译不知道能否成功。
哈哈

norikohxb, 2020-03-07:
编译不成功，悲剧了。不是简单的加上一个井号就可以了，虽然可以出现菜单，可以勾选。但是编译失败。黑脸

norikohxb, 2020-03-07:
rm -rf /home/vehxb/friendlywrt-h5/friendlywrt/package/lean/adbyby
rm -rf /home/vehxb/friendlywrt-h5/friendlywrt/package/lean/luci-app-adbyby-plus
cp -r /home/vehxb/friendlywrt-rk3328/friendlywrt/package/lean/luci-app-adbyby-plus /home/vehxb/friendlywrt-h5/friendlywrt/package/lean/
cp -r /home/vehxb/friendlywrt-rk3328/friendlywrt/package/lean/adbyby /home/vehxb/friendlywrt-h5/friendlywrt/package/lean/

用几天前没有更新的adbyby替换新的了，其实早上还可以
就是自己手贱，天天去git pull 源代码，就希望保持最新

iyuangang, 2020-03-07:
> 编译不成功，悲剧了。不是简单的加上一个井号就可以了，虽然可以出现菜单，可以勾选。但是编译失败。黑脸

改成这个
[adbyby version 20181023](https://raw.githubusercontent.com/iyuangang/openwrt/master/package/lean/adbyby/Makefile)

norikohxb, 2020-03-07:
我还好硬盘里面还有几天前的adbyby。用了可以了。
现在是新的lean的的luci-app-ssr-plus好像也没有办法选v2ray，和shadowsock。
选了编译也不成功，

coolsnowwolf, 2020-03-07:
我有点疑惑，R1s 不是 ARMv8 64bit 的吗？ 能用？

lesliedh, 2020-03-08:
DEPENDS:=@(x86||x86_64||arm||mipsel||mips||aarch64)
ifeq ($(ARCH),aarch64)
	ADBYBY_DIR:=armv7
endif

norikohxb, 2020-03-08:
> 我有点疑惑，R1s 不是 ARMv8 64bit 的吗？ 能用？

不能用，到了昨天下午不能用
我比较有强迫症，一天git pull你的lede目录4次。
早上还可以，下午git pull的package 复制到h5 选择h5的型号就不见了。

norikohxb, 2020-03-08:
> DEPENDS:=@(x86||x86_64||arm||mipsel||mips||aarch64)
> ifeq ($(ARCH),aarch64)
> ADBYBY_DIR:=armv7
> endif

你的意思是把新的adbyby的Makefile的内容修改一下就可以用了吗？可以编译通过吗？
我昨天不懂得修改我就在depnds前面加上井号可以出现菜单，但是编译会失败。
你这样修改是不是编译就可以过了？

iyuangang, 2020-03-08:
> > DEPENDS:=@(x86||x86_64||arm||mipsel||mips||aarch64)
> > ifeq ($(ARCH),aarch64)
> > ADBYBY_DIR:=armv7
> > endif
> 
> 你的意思是把新的adbyby的Makefile的内容修改一下就可以用了吗？可以编译通过吗？
> 我昨天不懂得修改我就在depnds前面加上井号可以出现菜单，但是编译会失败。
> 你这样修改是不是编译就可以过了？

要是实在不会的话，可以用我的项目 [OpenWrt Stable 1907 with lean's package](https://github.com/iyuangang/openwrt) 
经过测试可以成功生成固件。
可以clone 到本地编译，也可以读一下 .github/workflows/openwrt-ci-v2.yml , 修改为你的设备。

lesliedh, 2020-03-08:
> > DEPENDS:=@(x86||x86_64||arm||mipsel||mips||aarch64)
> > ifeq ($(ARCH),aarch64)
> > ADBYBY_DIR:=armv7
> > endif
> 
> 你的意思是把新的adbyby的Makefile的内容修改一下就可以用了吗？可以编译通过吗？
> 我昨天不懂得修改我就在depnds前面加上井号可以出现菜单，但是编译会失败。
> 你这样修改是不是编译就可以过了？

可以编译通过

lesliedh, 2020-03-08:
> > DEPENDS:=@(x86||x86_64||arm||mipsel||mips||aarch64)
> > ifeq ($(ARCH),aarch64)
> > ADBYBY_DIR:=armv7
> > endif
> 
> 你的意思是把新的adbyby的Makefile的内容修改一下就可以用了吗？可以编译通过吗？
> 我昨天不懂得修改我就在depnds前面加上井号可以出现菜单，但是编译会失败。
> 你这样修改是不是编译就可以过了？

可以编译通过。

closed 2020-03-08 by coolsnowwolf

norikohxb, 2020-03-08:
谢谢你大佬们搞定了，lean大大将来再更新的adbyby能否直接兼容nanopi。毕竟我买了2个这种迷你设备，如果可以就比较省事。