coolsnowwolf/lede #3636: SSR Plus+更新订阅失败

fonlan, 2020-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：拉取今天最新的代码自编译的X64软路由版本，SSR Plus+里更新订阅的时候提示：
2020-03-07 19:29:36 更新失败，没有可用的节点信息
2020-03-07 19:29:36  https://subscribe.xxxx.com/api/v2ray/xxxxxxxxxxxxxxxxxxx: 获取内容为空
 https://subscribe.xxxx.com/api/v2ray/xxxxxxxxxxxxxxxxxxx: Scheme missing.

### (2) 路由器型号和固件版本

A：X64软路由

### (3) 详细日志

A：2020-03-07 19:29:36 更新失败，没有可用的节点信息
2020-03-07 19:29:36 更新失败，没有可用的节点信息
2020-03-07 19:29:36  https://subscribe.xxxx.com/api/v2ray/xxxxxxxxxxxxxxxxxxx: 获取内容为空
 https://subscribe.xxxx.com/api/v2ray/xxxxxxxxxxxxxxxxxxx: Scheme missing.
手动尝试wget-ssl https://subscribe.xxxx.com/api/v2ray/xxxxxxxxxxxxxxxxxxx可以正常返回内容

closed 2020-03-08 by coolsnowwolf

hhyygg2009, 2020-03-18:
我也有同样的问题 

OpenWrt R20.2.15 / LuCI Master (git-20.062.50551-a6189c5)
3月3号编译的固件

azhenge, 2020-04-05:
论坛的固件试了个遍，都有这个问题
更新失败，没有可用的节点信息

dongfenglong, 2020-05-26:
我的也是
OpenWrt R20.4.8