coolsnowwolf/lede #3639: 请教一个关于电脑开启SOCKS5端口，主路由器连接电脑socks5的问题

lizhenglin2002, 2020-03-07:
[image]
[image]
[image]
我在电脑v2ray开启了socks5，电脑做运算，主路由新3转发。但是这样设置网络很慢。请教一下应该怎样设置

hyird, 2020-03-07:
这不是论坛。

lizhenglin2002, 2020-03-07:
> 这不是论坛。
哦

lizhenglin2002, 2020-03-07:
> 这不是论坛。

哦

coolsnowwolf, 2020-03-07:
服务器 LAN IP 需要加入上级 OP 的LAN IP 白名单中

lizhenglin2002, 2020-03-07:
> 服务器LAN IP需要加入上级OP的LAN IP白名单中

[image]
请问是这样吗？

lizhenglin2002, 2020-03-07:
> 服务器 LAN IP 需要加入上级 OP 的LAN IP 白名单中

已解决，谢谢

closed 2020-03-07 by lizhenglin2002

piaoyizy, 2020-03-12:
> > 服务器LAN IP需要加入上级OP的LAN IP白名单中
> 
> 已解决，谢谢

请问 是 设置   不走代理的局域网LAN IP  吗， 我和你一样现在慢。

lizhenglin2002, 2020-03-12:
我已经放弃了，随然可以上网了，但还是很慢

piaoyizy, 2020-03-12:
> […](#)

是的，是很慢，不知道是不是设置有问题。