coolsnowwolf/lede #3641: lean大再请教您一下，最新的lede的package复制到nanopi H5去的问题

norikohxb, 2020-03-07:
以前ddns直接勾选了就行了，现在为什么还要勾选aliyun和dospod才可以编译成功？

以前ssrplus可以选择v2ray，现在如果在ssrplus里面选择v2ray编译就不通过？

我说的这些都是针对现在最新的lede，
大约一个星期以前的lede是没有这个问题。
能否在百忙中回答一下问题。谢谢了

iyuangang, 2020-03-07:
`SSR Plus +` 需要 `tools`下的 `ucl` `upx` 两个包，还需要改`tools/Makefile` 
你自己看看L大的代码研究一下

norikohxb, 2020-03-07:
看哪个文档？我的trojan可以在r1s编译通过，但是如果还要勾选v2ray就编译失败了。同样之前的版本不会，也还是自己天天希望把lean的package保持最新，
是luci-app-ssr-plus里面的Makefile对吗？

norikohxb, 2020-03-07:
> `SSR Plus +` 需要 `tools`下的 `ucl` `upx` 两个包，还需要改`tools/Makefile`
> 你自己看看L大的代码研究一下
能否再具体一点点，我现在是想要吧ssr-plus里面的子选项都打钩上去。

norikohxb, 2020-03-07:
是lede/tools 对吧？
还有upx，这两个因为没有一起复制到friendlywrt目录去所以造成问题？

iyuangang, 2020-03-07:
> 是lede/tools 对吧？
> 还有upx，这两个因为没有一起复制到friendlywrt目录去所以造成问题？

是的，然后在 friendlywrt 目录 tools/Makefile 添加 
```
tools-y += ucl upx
$(curdir)/upx/compile := $(curdir)/ucl/compile
```

norikohxb, 2020-03-07:
这个是针对ssr-plus那么再请教一下，现在最新的unblock netease好像也编译不了
是又有什么玄机，lean大多了golang
[image]

norikohxb, 2020-03-07:
唉感觉我这种不懂得修改代码的，还天天要最新，追着lean大的最新程序跑，在x86平台是很轻松，lean大都给你搞好了，在arm平台，最近几次更新尝到苦头了，刚购买r1s那几天编译还不会，怎么更新怎么编译都行。

closed 2020-03-08 by coolsnowwolf