coolsnowwolf/lede #3642: 编译v2ray出错!

jeasonliang, 2020-03-07:
1.关于要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
没有类似的issue

 2. 详细叙述
各位好!我是看了油管上的教程突发奇想想把lede的ssr plus移植到原版的openwrt上去!我的步骤是这样的:
在VPS上编译,不存在网络给墙的问题

git clone https://github.com/openwrt/openwrt
git clone https://github.com/coolsnowwolf/lede
cd lede
git pull
 ./scripts/feeds update -a 
 ./scripts/feeds install -a
一切正常,然后我把lede下面的package里的lede拷贝到openwrt对应的位置里去
cd ~/lede/package
cp -r lean/ ~/openwrt/package/
同时把lede下面feeds/packages/lang/golang里的拷贝到openwrt相对应位置去
cp -r golang/* ~/openwrt/feeds/packages/lang/golang/
然后切换到openwrt目录
cd ~/openwrt
执行git pull
 ./scripts/feeds update -a 
 ./scripts/feeds install -a
选择对应的芯片及型号 make menuconfig 选择好相对应的一切
make -j1 V=s开始编译

3:出现的具体问题

A：如下: 在编译v2ray时候出现错误~!
[image]

Building targets

Cleaning module download cache (golang/go#27455)

mv -f /home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_mips/main /home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_mips/v2ray
/home/vm/openwrt/staging_dir/host/bin/upx --lzma --best /home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_mips/v2ray
bash: /home/vm/openwrt/staging_dir/host/bin/upx: No such file or directory
Makefile:307: recipe for target '/home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.built' failed
make[3]: *** [/home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.built] Error 127
make[3]: Leaving directory '/home/vm/openwrt/package/lean/v2ray'
time: package/lean/v2ray/compile#7.44#5.03#10.40
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/vm/openwrt'
package/Makefile:107: recipe for target '/home/vm/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/vm/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vm/openwrt'
/home/vm/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
vm@ddrdff:~/openwrt$ df
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1268144        0   1268144   0% /dev
tmpfs             256132    27840    228292  11% /run
/dev/vda1       38672000 18537148  18168884  51% /
tmpfs            1280656        0   1280656   0% /dev/shm
tmpfs               5120        0      5120   0% /run/lock
tmpfs            1280656        0   1280656   0% /sys/fs/cgroup
tmpfs             256128        0    256128   0% /run/user/1000

(2) 路由器型号和固件版本

A：网件wndr3800

 (3) 详细日志

A：如下
go: downloading google.golang.org/grpc v1.24.0
go: downloading github.com/golang/protobuf v1.3.2
go: downloading golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: downloading golang.org/x/sys v0.0.0-20190215142949-d0b11bdaac8a
go: downloading golang.org/x/net v0.0.0-20190311183353-d8887717615a
go: downloading github.com/refraction-networking/utls v0.0.0-20190909200633-43c36d3c1f57
go: downloading google.golang.org/genproto v0.0.0-20180831171423-11092d34479b
go: downloading github.com/gorilla/websocket v1.4.1
go: downloading go.starlark.net v0.0.0-20190919145610-979af19b165c
go: downloading golang.org/x/text v0.3.0

Building targets

Cleaning module download cache (golang/go#27455)

mv -f /home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_mips/main /home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_mips/v2ray
/home/vm/openwrt/staging_dir/host/bin/upx --lzma --best /home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.go_work/build/bin/linux_mips/v2ray
bash: /home/vm/openwrt/staging_dir/host/bin/upx: No such file or directory
Makefile:307: recipe for target '/home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.built' failed
make[3]: *** [/home/vm/openwrt/build_dir/target-mips_24kc_musl/v2ray-core-4.22.1/.built] Error 127
make[3]: Leaving directory '/home/vm/openwrt/package/lean/v2ray'
time: package/lean/v2ray/compile#7.44#5.03#10.40
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/vm/openwrt'
package/Makefile:107: recipe for target '/home/vm/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/vm/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vm/openwrt'
/home/vm/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
 请各位指引一下问题在什么地方!

iyuangang, 2020-03-07:
`v2ray` 缺少 `upx` `ucl` 
1. 下载官方 stable release 下的 [最新稳定版本](https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz)
`wget https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz`
2. 将L大的`package/lean` 拷贝到 上面的package 下
3. 将L大的`tools/upx` 和 `tools/ucl` 拷贝到上面的 tools 下
4. 修改 `tools` 下的 `Makefile` 添加两行
```
tools-y += ucl upx
$(curdir)/upx/compile := $(curdir)/ucl/compile
```
5. 开始编译
[成功案例](https://github.com/iyuangang/openwrt/issues/2)

jeasonliang, 2020-03-08:
> `v2ray` 缺少 `upx` `ucl`
> 
> 1. 下载官方 stable release 下的 [最新稳定版本](https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz)
>    `wget https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz`
> 2. 将L大的`package/lean` 拷贝到 上面的package 下
> 3. 将L大的`tools/upx` 和 `tools/ucl` 拷贝到上面的 tools 下
> 4. 修改 `tools` 下的 `Makefile` 添加两行
> 
> ```
> tools-y += ucl upx
> $(curdir)/upx/compile := $(curdir)/ucl/compile
> ```
> 
> 1. 开始编译
>    [成功案例](https://github.com/iyuangang/openwrt/issues/2)

感谢🙏！谢谢🙏！

closed 2020-03-08 by coolsnowwolf

adamgonglj, 2020-03-10:
> 感谢🙏！谢谢🙏！

是不是就编译成功了？

jeasonliang, 2020-03-13:
> > > `v2ray`·`upx` `ucl`
> > > 
> > > 1. • 稳定释放[|](https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz)
> > >    `wget https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz`
> > > 2. [L] 包`package/lean`
> > > 3. [ 工具 ]`tools/upx``tools/ucl`
> > > 4. •`tools``Makefile`
> > > 
> > > ```
> > > tools-y += ucl upx
> > > $(curdir)/upx/compile := $(curdir)/ucl/compile
> > > ```
> > > 
> > > 
> > > 
> > > 1. [•](https://github.com/iyuangang/openwrt/issues/2)
> > 
> > 
> > pray！！！pray
> 
> ？

> > > `v2ray` 缺少 `upx` `ucl`
> > > 
> > > 1. 下载官方 stable release 下的 [最新稳定版本](https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz)
> > >    `wget https://github.com/openwrt/openwrt/archive/v19.07.2.tar.gz`
> > > 2. 将L大的 拷贝到 上面的package 下`package/lean`
> > > 3. 将L大的 和  拷贝到上面的 tools 下`tools/upx``tools/ucl`
> > > 4. 修改  下的  添加两行`tools``Makefile`
> > > 
> > > ```
> > > tools-y += ucl upx
> > > $(curdir)/upx/compile := $(curdir)/ucl/compile
> > > ```
> > > 
> > > 
> > > 
> > > 1. 开始编译
> > >    [成功案例](https://github.com/iyuangang/openwrt/issues/2)
> > 
> > 
> > 感谢pray！谢谢pray！
> 
> 是不是就编译成功了？

这段时间忙着 没时间搞

Dboykey, 2020-07-01:
我刚刚自己硬啃过了这个问题，才发现这个帖子……好想哭……