coolsnowwolf/lede #3645: SSR PLUS节点会偶尔中断，节点添加选项中有一行向上向下箭头图标显示缺少

weweyes, 2020-03-08:
SSR PLUS节点会偶尔中断，节点添加选项中有一行向上向下箭头图标显示缺少

iyuangang, 2020-03-08:
1. 节点中断原因很多，机场稳定性，自建服务端稳定性，isp 中断，防火墙识别特征阻滞等。
2. 问题2的截图呢？是什么主题？

weweyes, 2020-03-08:
问题2：节点添加选项中有一行向上向下箭头图标显示缺少，VMware ESXI  workstation测试过都出现同样问题
[image]

weweyes, 2020-03-08:
> 1. 节点中断原因很多，机场稳定性，自建服务端稳定性，isp 中断，防火墙识别特征阻滞等。
> 2. 问题2的截图呢？是什么主题？

问题2：节点添加选项中有一行向上向下箭头图标显示缺少，VMware ESXI workstation测试过都出现同样问题
[image]

closed 2020-03-08 by coolsnowwolf