coolsnowwolf/lede #3647: ac58u内存迅雷全速下载内存耗尽

ggllioio, 2020-03-08:
昨天3.7日编译的，已经去掉了samba，adbyby，ssr。增加了ipv6。家里100m电信宽带pppoe，开迅雷下载超不过20s就会内存耗尽断网。后重新编译开启zram情况有好转。但手机app提速200m后依然会内存耗尽断网。几个月前的版本应该没问题。望修复。
附上日志
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.628624] Hardware name: Generic DT based system
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.634904] [<c030e45c>] (unwind_backtrace) from [<c030a968>] (show_stack+0x10/0x14)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.639663] [<c030a968>] (show_stack) from [<c0764634>] (dump_stack+0x94/0xa8)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.647561] [<c0764634>] (dump_stack) from [<c03abac8>] (dump_header+0x9c/0x1fc)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.654589] [<c03abac8>] (dump_header) from [<c03aad70>] (oom_kill_process+0xc8/0x580)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.662140] [<c03aad70>] (oom_kill_process) from [<c03ab9b4>] (out_of_memory+0x310/0x334)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.669868] [<c03ab9b4>] (out_of_memory) from [<c03afc50>] (__alloc_pages_nodemask+0xabc/0xe94)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.678120] [<c03afc50>] (__alloc_pages_nodemask) from [<c03e3bec>] (__read_swap_cache_async+0xa4/0x1e8)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.686630] [<c03e3bec>] (__read_swap_cache_async) from [<c03e42f0>] (do_swap_page_readahead+0x80/0x140)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.696352] [<c03e42f0>] (do_swap_page_readahead) from [<c03d2dfc>] (do_swap_page+0x78/0x580)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.705810] [<c03d2dfc>] (do_swap_page) from [<c03d4138>] (handle_mm_fault+0xa9c/0xbfc)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.714227] [<c03d4138>] (handle_mm_fault) from [<c030f1c4>] (do_page_fault+0x128/0x2ac)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.722039] [<c030f1c4>] (do_page_fault) from [<c03012a0>] (do_DataAbort+0x3c/0xb4)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.730391] [<c03012a0>] (do_DataAbort) from [<c030b81c>] (__dabt_usr+0x3c/0x40)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.737742] Exception stack(0xc27b1fb0 to 0xc27b1ff8)
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.745380] 1fa0:                                     b6f05840 00000300 00000300 00000032
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.750336] 1fc0: b6f05840 bed19aec 0000143c b6f05df0 b6bf6ec0 b6bf6e30 00110a08 0006bf38
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.758491] 1fe0: 00000000 bed19ae8 000b07b8 0007bff4 60000010 ffffffff
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.766683] Mem-Info:
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.773088] active_anon:0 inactive_anon:2 isolated_anon:0
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.773088]  active_file:170 inactive_file:187 isolated_file:0
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.773088]  unevictable:0 dirty:0 writeback:0 unstable:0
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.773088]  slab_reclaimable:684 slab_unreclaimable:3197
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.773088]  mapped:15 shmem:0 pagetables:115 bounce:0
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.773088]  free:3383 free_pcp:51 free_cma:0
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.784723] Node 0 active_anon:0kB inactive_anon:8kB active_file:680kB inactive_file:748kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:60kB dirty:0kB writeback:0kB shmem:0kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.806995] Normal free:13596kB min:16384kB low:20480kB high:24576kB active_anon:0kB inactive_anon:8kB active_file:720kB inactive_file:748kB unevictable:0kB writepending:0kB present:129024kB managed:120856kB mlocked:0kB kernel_stack:936kB pagetables:460kB bounce:0kB free_pcp:204kB local_pcp:16kB free_cma:0kB
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.833580] lowmem_reserve[]: 0 0 0
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.855812] Normal: 155*4kB (UME) 296*8kB (UM) 271*16kB (UM) 67*32kB (UM) 18*64kB (UM) 13*128kB (UM) 3*256kB (UM) 1*512kB (U) 0*1024kB 0*2048kB 0*4096kB = 13564kB
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.859112] 369 total pagecache pages
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.873695] 1 pages in swap cache
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.877426] Swap cache stats: add 9750, delete 9749, find 713/8470
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.880711] Free swap  = 56316kB
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.886800] Total swap = 60412kB
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.890175] 32256 pages RAM
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.894363] 0 pages HighMem/MovableOnly
Sun Mar  8 12:23:14 2020 kern.warn kernel: [  411.895910] 2042 pages reserved
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.899720] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.902878] [  251]     0   251      252        2       5       0       18             0 ubusd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.911581] [  273]     0   273      176        0       4       0        9             0 askfirst
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.920071] [  288]     0   288      204        0       4       0       11             0 urngd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.929029] [ 1445]     0  1445      258        2       4       0       46             0 logd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.937520] [ 1466]     0  1466      337        2       4       0       26             0 rpcd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.946128] [ 1782]     0  1782      271        2       4       0       10             0 crond
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.954619] [ 2272]     0  2272      248        0       3       0       14             0 vsftpd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.963104] [ 3371]     0  3371      210        0       4       0       14             0 dropbear
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.971716] [ 3718]     0  3718      281        2       5       0       27             0 nlbwmon
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.980740] [ 3843]     0  3843      183        0       3       0        9             0 vlmcsd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.989596] [ 5413]     0  5413      271        0       4       0       22             0 ntpd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  411.998023] [ 7408]   453  7408      298        0       5       0       29             0 dnsmasq
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.006700] [ 7417]     0  7417      226        0       6       0       25             0 dropbear
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.015565] [ 7436]     0  7436      389        2       4       0       45             0 netifd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.024326] [ 7437]     0  7437      320        2       4       0       25             0 odhcpd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.032810] [ 7438]     0  7438      336        0       5       0       26             0 uhttpd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.041511] [ 7505]     0  7505      210        0       3       0       14             0 odhcp6c
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.050192] [ 7513]     0  7513      270        0       4       0        9             0 udhcpc
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.059214] [ 8334]     0  8334     1173        0       6       0       78             0 hostapd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.067641] [ 8380]     0  8380     1173        0       7       0       73             0 hostapd
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.076658] [11078]     0 11078      226        0       6       0       25             0 dropbear
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.085431] [11080]     0 11080      272        0       4       0       12             0 ash
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.094198] [11090]     0 11090      554        0       5       0        1             0 htop
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.102686] [11145]     0 11145      420        0       4       0       27             0 fw3
Sun Mar  8 12:23:14 2020 kern.info kernel: [  412.111137] [11149]     0 11149      242        0       3       0        7             0 dhcpv6.script
[... cut ...]
Sun Mar  8 12:23:28 2020 daemon.notice netifd: radio0 (11192): command failed: Not supported (-95)
Sun Mar  8 12:23:28 2020 daemon.notice netifd: radio1 (11194): command failed: Not supported (-95)
Sun Mar  8 12:23:28 2020 user.notice mac80211: Failed command: iw phy phy0 set distance 0
Sun Mar  8 12:23:28 2020 user.notice mac80211: Failed command: iw phy phy1 set distance 0
Sun Mar  8 12:23:29 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Sun Mar  8 12:23:29 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf

ggllioio, 2020-03-08:
日志状态是：58u串联在主路由下，wan口dhcp。内存耗尽时ssh也会一并杀死。

ggllioio, 2020-03-08:
内核日志：

> [  243.435499] Hardware name: Generic DT based system
[  243.441694] [<c030e45c>] (unwind_backtrace) from [<c030a968>] (show_stack+0x10/0x14)
[  243.446369] [<c030a968>] (show_stack) from [<c0764634>] (dump_stack+0x94/0xa8)
[  243.454261] [<c0764634>] (dump_stack) from [<c03abac8>] (dump_header+0x9c/0x1fc)
[  243.461289] [<c03abac8>] (dump_header) from [<c03aad70>] (oom_kill_process+0xc8/0x580)
[  243.468844] [<c03aad70>] (oom_kill_process) from [<c03ab9b4>] (out_of_memory+0x310/0x334)
[  243.476571] [<c03ab9b4>] (out_of_memory) from [<c03afc50>] (__alloc_pages_nodemask+0xabc/0xe94)
[  243.484821] [<c03afc50>] (__alloc_pages_nodemask) from [<c0314ff4>] (copy_process.part.3+0x104/0x1614)
[  243.493326] [<c0314ff4>] (copy_process.part.3) from [<c0316660>] (_do_fork+0xbc/0x360)
[  243.502699] [<c0316660>] (_do_fork) from [<c031697c>] (sys_fork+0x24/0x2c)
[  243.510595] [<c031697c>] (sys_fork) from [<c0307680>] (ret_fast_syscall+0x0/0x54)
[  243.517488] Mem-Info:
[  243.525032] active_anon:0 inactive_anon:0 isolated_anon:0
[  243.525032]  active_file:116 inactive_file:175 isolated_file:0
[  243.525032]  unevictable:0 dirty:0 writeback:0 unstable:0
[  243.525032]  slab_reclaimable:606 slab_unreclaimable:3156
[  243.525032]  mapped:2 shmem:0 pagetables:102 bounce:0
[  243.525032]  free:4400 free_pcp:1 free_cma:0
[  243.536132] Node 0 active_anon:0kB inactive_anon:0kB active_file:464kB inactive_file:700kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:8kB dirty:0kB writeback:0kB shmem:0kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
[  243.558394] Normal free:17632kB min:16384kB low:20480kB high:24576kB active_anon:0kB inactive_anon:0kB active_file:464kB inactive_file:700kB unevictable:0kB writepending:0kB present:129024kB managed:120856kB mlocked:0kB kernel_stack:888kB pagetables:408kB bounce:0kB free_pcp:4kB local_pcp:0kB free_cma:0kB
[  243.584726] lowmem_reserve[]: 0 0 0
[  243.606955] Normal: 227*4kB (UMH) 191*8kB (UMEH) 152*16kB (UMEH) 70*32kB (UMH) 51*64kB (UMEH) 23*128kB (UMEH) 5*256kB (UM) 2*512kB (U) 0*1024kB 1*2048kB (E) 0*4096kB = 17668kB
[  243.610338] 300 total pagecache pages
[  243.625966] 0 pages in swap cache
[  243.629764] Swap cache stats: add 4417, delete 4417, find 293/3430
[  243.633082] Free swap  = 56572kB
[  243.639139] Total swap = 60412kB
[  243.642544] 32256 pages RAM
[  243.645734] 0 pages HighMem/MovableOnly
[  243.648253] 2042 pages reserved
[  243.652073] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[  243.655248] [  251]     0   251      252        0       5       0       34             0 ubusd
[  243.663914] [  273]     0   273      176        0       4       0        9             0 askfirst
[  243.672420] [  288]     0   288      204        0       4       0       11             0 urngd
[  243.681345] [ 1445]     0  1445      258        0       4       0       53             0 logd
[  243.689872] [ 1466]     0  1466      337        2       4       0       26             0 rpcd
[  243.698465] [ 1678]     0  1678      389        0       6       0       67             0 netifd
[  243.706969] [ 1715]     0  1715      313        0       5       0       43             0 odhcpd
[  243.715473] [ 1782]     0  1782      271        2       4       0       11             0 crond
[  243.724155] [ 1825]     0  1825      270        0       4       0        9             0 udhcpc
[  243.732834] [ 1854]     0  1854      210        0       4       0       27             0 odhcp6c
[  243.741413] [ 2135]     0  2135      342        0       5       0       34             0 uhttpd
[  243.750459] [ 2272]     0  2272      248        0       3       0       14             0 vsftpd
[  243.758877] [ 3371]     0  3371      210        0       4       0       10             0 dropbear
[  243.767560] [ 3718]     0  3718      281        2       5       0       29             0 nlbwmon
[  243.776586] [ 3843]     0  3843      183        0       3       0        9             0 vlmcsd
[  243.785441] [ 5413]     0  5413      271        0       4       0       24             0 ntpd
[  243.793859] [ 7382]     0  7382      249        1       5       0       39             0 luci
[  243.802546] [ 7393]     0  7393      329        0       5       0       90             0 sh
[  243.811030] [ 7394]     0  7394      329        1       5       0       80             0 sh
[  243.819206] [ 7395]     0  7395      308        0       4       0       72             0 sh
[  243.827541] [ 7406]     0  7406      286        0       4       0       24             0 ubus
[  243.835872] [ 7407]     0  7407      269        0       3       0        5             0 ubus
[  243.844555] [ 7408]     0  7408      199        0       3       0        7             0 dnsmasq
[  243.853053] Out of memory: Kill process 7393 (sh) score 2 or sacrifice child
[  243.861911] Killed process 7393 (sh) total-vm:1316kB, anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
[  243.869854] ath10k_ahb a000000.wifi: SWBA overrun on vdev 0, skipped old beacon
[  243.884276] uhttpd invoked oom-killer: gfp_mask=0x15080c0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), nodemask=(null),  order=1, oom_score_adj=0
[  243.884711] uhttpd cpuset=/ mems_allowed=0
[  243.897417] CPU: 2 PID: 2135 Comm: uhttpd Not tainted 4.14.171 #0
[  243.900690] Hardware name: Generic DT based system
[  243.906886] [<c030e45c>] (unwind_backtrace) from [<c030a968>] (show_stack+0x10/0x14)
[... cut ...]
[  443.187043] br-lan: port 3(wlan1) entered disabled state
[  443.192839] device wlan1 entered promiscuous mode
[  443.416478] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  443.417076] br-lan: port 2(wlan0) entered blocking state
[  443.422089] br-lan: port 2(wlan0) entered forwarding state
[  443.707464] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  443.707826] br-lan: port 3(wlan1) entered blocking state
[  443.712982] br-lan: port 3(wlan1) entered forwarding state

lilunxm12, 2020-03-08:
ac58u官方就有oom的报告,openwrt官方推荐的做法是换原装高通ath10k-ct驱动或者用smallbuffers版的ath10k-ct

coolsnowwolf, 2020-03-08:
硬改内存即可

ggllioio, 2020-03-08:
> ac58u官方就有oom的报告，openwrt官方推荐的做法是换原装高通ath10k-ct驱动或者用smallbuffers版的ath10k-ct

感谢~~
又重新编译使用ath10k-firmware-qca4019-ct和kmod-ath10k-ct-smallbuffers成功！重新测试完全正常。200m环境也毫无压力。
开心再次感谢~~

ggllioio, 2020-03-08:
> 硬改内存即可

硬改实在没那个技术。。。已经换成ath10k-firmware-qca4019-ct和kmod-ath10k-ct-smallbuffers。现在正常了。看了下官方源码用的就是这两个。可以考虑替换下。

closed 2020-03-08 by ggllioio

lilunxm12, 2020-03-08:
> > 硬改内存即可
> 
> 硬改实在没那个技术。。。已经换成ath10k-firmware-qca4019-ct和kmod-ath10k-ct-smallbuffers。现在正常了。看了下官方源码用的就是这两个。可以考虑替换下。

....其实我打错了,应该是换成不带ct的版本或者smallbuffer的ct版.openwrt从19.07开始换成默认用ct版

ggllioio, 2020-03-08:
> > > 硬改内存即可
> > 
> > 
> > 硬改实在没那个技术。。。已经换成ath10k-firmware-qca4019-ct和kmod-ath10k-ct-smallbuffers。现在正常了。看了下官方源码用的就是这两个。可以考虑替换下。
> 
> ....其实我打错了,应该是换成不带ct的版本或者smallbuffer的ct版.openwrt从19.07开始换成默认用ct版

？？？迷惑。默认爆内存的就是不带ct版本。我看了官方就是现在这两个。现在一切正常。