coolsnowwolf/lede #3656: 1、3.8编译失败，显示mac80211.ko文件丢失； 2，特定情况下qBittorrent与SSR +冲突，且无法恢复

vodott, 2020-03-08:
## 1.关于你要提交的问题
问：是否搜索了issue（使用“ x”选择）
[x]没有类似的问题

## 2. 详细叙述

### (1) 具体问题
*今天git pull后编译

a.编译失败，似乎包里少了mac80211.ko这个文件

b.其实之所以重新编译固件是因为qBittorrent与SSR+发生冲突，崩溃后再也没有能运行过。
我的路由是同时开启ssr+ 与qBittorrent，本来一直没问题，但当我从qBittorrent外挂硬盘中往pc拖大文件（电影）时（此时网速达到110mb/s（此时家里千兆网络已满速），路由器cpu也是满负荷），SSR+直接崩溃（无法科学上网也无法连接国内ip，停用后网络恢复正常，再启用还是无法科学上网，重启路由也不行），在我关闭qBittorrent（取消“启用”）后SSR+恢复正常。可是当我再次“启用”qBittorrent时，无论如何（重启、重新刷新版本[3.4号编译]固件都不行。之后我刷了原先的固件（2.23编译），“启用”qBittorrent后显示webui正常【但无法登录，说ip登录太多次被封，这个有点莫名其秒】，但ssr+此时又不能使用了，关闭qBittorrent后，ssr+正常，重新启用qBittorrent后，qBittorrent的webui再也打不开）也无法启动webui，我开启qBittorrent日志功能，但在目录中也找不到日志文件，在系统进程里也找不到与qBittorrent相关的进程。

### (2) 路由器型号和固件版本

主机型号 | Linksys WRT3200ACM
架构 | ARMv7 Processor rev 1 (v7l) x 2
固件版本 | OpenWrt R20.3.5 / LuCI Master (git-20.062.50551-a6189c5)
内核版本 | 4.19.106

### (3) 详细日志

make[3]: Entering directory `/home/zhangqi/lede/package/kernel/mac80211'
removed '/home/zhangqi/lede/bin/targets/mvebu/cortexa9/packages/kmod-mac80211_4.19.106+4.19.98-1-1_arm_cortex-a9_vfpv3.ipk'
mkdir -p /home/zhangqi/lede/bin/targets/mvebu/cortexa9/packages /home/zhangqi/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/backports-4.19.98-1/ipkg-arm_cortex-a9_vfpv3/kmod-mac80211/CONTROL /home/zhangqi/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/pkginfo
ERROR: module '/home/zhangqi/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/backports-4.19.98-1/net/mac80211/mac80211.ko' is missing.
make[3]: *** [/home/zhangqi/lede/bin/targets/mvebu/cortexa9/packages/kmod-mac80211_4.19.106+4.19.98-1-1_arm_cortex-a9_vfpv3.ipk] Error 1
make[3]: Leaving directory `/home/zhangqi/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.05#0.02#0.16
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/home/zhangqi/lede'
make[1]: *** [/home/zhangqi/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/zhangqi/lede'
make: *** [world] Error 2

lilunxm12, 2020-03-08:
我也遇到相同问题

vodott, 2020-03-08:
> 我也遇到相同问题

您是指第1个问题还是第2个问题？

flyfeel, 2020-03-08:
同mac80211报错！！

ricksuzade-maker, 2020-03-08:
time: package/network/services/hostapd/wpad-full-internal/compile#0.92#0.35#1.27
make[3]: Entering directory '/home/runner/work/lede/lede/package/network/utils/iw'
mkdir -p /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/iw-tiny/iw-4.14/.pkgdir/iw/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.iw_installed
echo "iw" >> /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo/iw.tiny.install
make[3]: Leaving directory '/home/runner/work/lede/lede/package/network/utils/iw'
time: package/network/utils/iw/tiny/compile#0.12#0.02#0.14
make[3]: Entering directory '/home/runner/work/lede/lede/package/kernel/mac80211'
mkdir -p /home/runner/work/lede/lede/bin/targets/x86/64/packages /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1/ipkg-x86_64/kmod-mac80211/CONTROL /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.98-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/runner/work/lede/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.106+4.19.98-1-1_x86_64.ipk' failed
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/mac80211'
make[3]: *** [/home/runner/work/lede/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.106+4.19.98-1-1_x86_64.ipk] Error 1
time: package/kernel/mac80211/compile#0.20#0.05#0.24
package/Makefile:111: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[2]: *** [package/kernel/mac80211/compile] Error 2
package/Makefile:107: recipe for target '/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
##[error]Process completed with exit code 2.

ricksuzade-maker, 2020-03-08:
我刚刚也遇到同样的问题了，不知道咋回事

flyfeel, 2020-03-08:
大雕 请把mac80211驱动下架 谢谢
4天前有人把这个驱动加上去了。。。。

1715173329, 2020-03-08:
mac80211 is working well.

x86_64 mac80211 complilation (based on commit 2ae4662)
[image]

ipq40xx mac80211 complilation (based on commit 2ae4662)
[image]

dolphinpaopao, 2020-03-08:
目前看到问题的出现频次，x86似乎是问题最少最稳定的平台。

tone88, 2020-03-08:
同样的问题，正在用默认.config跑，希望能过

lilunxm12, 2020-03-08:
建议修改标题然后把问题2移到另外的issue,这下面都是说问题1的

vodott, 2020-03-08:
> 建议修改标题然后把问题2移到另外的问题，这下面都是说问题1的

如你所愿 ;-)

closed 2020-03-08 by coolsnowwolf

lilunxm12, 2020-03-08:
@coolsnowwolf 无意冒犯,但是三个小时之内有五个人出现同样问题,这样直接关闭是否不妥?还是说你也试过了没问题?

EstFoxs, 2020-03-08:
> @coolsnowwolf 无意冒犯,但是三个小时之内有五个人出现同样问题,这样直接关闭是否不妥?还是说你也试过了没问题?
已经正常编译了