coolsnowwolf/lede #3657: OpenVPN Server功能只监听ipv4,没有在ipv6监听 ?

you-mu, 2020-03-08:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：OpenVPN Server功能启用后,端口只在ipv4监听,没有在ipv6监听?

### (2) 路由器型号和固件版本

A：x86  /   (git-20.062.50551-a6189c5) / OpenWrt R20.3.5 

### (3) 详细日志

A：
[image]

segadrive, 2020-03-09:
是啊，没有ipv6，着急。如何监听IPV6

segadrive, 2020-03-09:
希望有大佬能解决这个问题

you-mu, 2020-03-09:
softetherVPN 的 openvpn 也会出现一致的情况.

lunatickochiya, 2020-03-09:
这个要先升级openvpn主程序后 修改luci 可以支持的

you-mu, 2020-03-09:
> 
> 
> 这个要先升级openvpn主程序后 修改luci 可以支持的

感谢大佬升级.
(git-20.062.50551-a6189c5) / OpenWrt R20.3.8  openvpn server已测试 , ipv6正常监听.

想请教一下, softetherVPN 监听为啥也只监听ipv4 ?

closed 2020-03-10 by coolsnowwolf