coolsnowwolf/lede #3660: 移动端打开web管理页面的2个问题

KnowledegAndAction, 2020-03-08:
## 1.Android手机端无法登录Lede管理页面

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Android手机浏览器打开Lede页面，登录界面输入用户名密码后点login，无法登录，没有任何提示。试了chrome/vivaldi/firefox都一样结果。但是这个问题是突然出现的，以前没有问题，可以正常登录。

### (2) 路由器型号和固件版本

A：x86 以及OpenWrt R9.12.31 / LuCI Master (git-19.338.43082-9e87e69)

## 2.iOS设备上用浏览器打开web管理页面SSRP的Server Nodes下不显示延迟，socket 连接情况显示connecting(实际在PC端浏览器看是connected)

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如图（chrome/safari一样结果）
https://i.loli.net/2020/03/08/qpcLH9Gtln8uCOy.png

### (2) 路由器型号和固件版本

A：x86 以及OpenWrt R9.12.31 / LuCI Master (git-19.338.43082-9e87e69)

coolsnowwolf, 2020-03-08:
直接用 firefox 即可

closed 2020-03-08 by coolsnowwolf

KnowledegAndAction, 2020-03-08:
已经列出firefox有一样问题，问题没有得到解决。

KnowledegAndAction, 2020-03-09:
> 直接用 firefox 即可

已经列出firefox有一样问题，问题没有得到解决。