coolsnowwolf/lede #3662: N2600的cpufreq调节不生效，一直保持最大频率

mjyhj, 2020-03-08:
CPU：N2600
工作模式选择高性能 OR 省电
待机频率选择600000
最大频率选择从600000到1600000
CPU触发频率，无论选择50% OR 99%

最终结果 CPU频率一直波动在50%左右，降了后马上又会升回来
CPU频率也是保持在1590+，降了马上又升回来

无论怎么调节，结果都是一样稳定在  CPU50%  1590Mhz

[image]
[image]

ja5onwu, 2020-03-08:
[image]

我的j1900连自定义都没

mjyhj, 2020-03-08:
> [image]
> 
> 我的j1900连自定义都没

我也只有2个

coolsnowwolf, 2020-03-08:
估计自动匹配到 Powersave 模式去了

Wilson-coder, 2020-03-09:
我的CPU是赛扬3205U，最大和最小频率都是锁定在1000。从概况页面能看到CPU的实际工作频率大约在500MHz~1.4xGHz之间波动。

closed 2020-03-09 by coolsnowwolf