coolsnowwolf/lede #3668: x86和N270编译问题

caissonfiv, 2020-03-08:
由于feeds/packages/lang/node更新到v12.15，x86和n270编译失败，也许是node-v12.15不支持x86的原因，有解决的办法吗？

coolsnowwolf, 2020-03-08:
是可以的，看首页

caissonfiv, 2020-03-08:
x86可以，N270就是过不了，应该不是缺少依赖的问题吧？

caissonfiv, 2020-03-08:
/home/caisson/backup/lede/package/feeds/packages/node
/home/caisson/backup/lede/package/feeds/packages/node-arduino-firmata
/home/caisson/backup/lede/package/feeds/packages/node-cylon
/home/caisson/backup/lede/package/feeds/packages/node-hid
/home/caisson/backup/lede/package/feeds/packages/node-mozilla-iot-gateway
/home/caisson/backup/lede/package/feeds/packages/node-serialport

caissonfiv, 2020-03-08:
/home/caisson/caissonfiv/lede/staging_dir/toolchain-i386_core2_gcc-7.5.0_musl/lib/gcc/i486-openwrt-linux-musl/7.5.0/../../../../i486-openwrt-linux-musl/bin/ld: node_task_queue.cc:(.text+0x788): undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
make[5]: *** [/home/caisson/caissonfiv/lede/build_dir/target-i386_core2_musl/node-v12.15.0/out/Release/node_mksnapshot] Error 1
rm f340e41b0030c3d6c67811d16509971ec59a98e8.intermediate 4582268a5471d3ed204baf17cdcbee07be9e4657.intermediate e73e6925efe5cd65e5dcc45ab3aed9d8653dee42.intermediate ce2cc396f8adefc8ff7fedcd986ed4cb6c6edd66.intermediate 374bf12ec5b26ae69cd17530fc7a513b4398cdf6.intermediate
make[5]: Leaving directory `/home/caisson/caissonfiv/lede/build_dir/target-i386_core2_musl/node-v12.15.0/out'
make[4]: *** [node] Error 2

coolsnowwolf, 2020-03-08:
工具链问题

caissonfiv, 2020-03-08:
能修复吗

mjyhj, 2020-03-09:
> 能修复吗

sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib

closed 2020-03-09 by coolsnowwolf