coolsnowwolf/lede #3671: 最新版本编译报错, 与node有关

zjc-enigma, 2020-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
clone了最新的源码
严格按照步骤执行编译, 无法通过
编译平台是ubuntu 16.04 amd64 LTS的VPS
VPS在国外, 不存在墙的问题
VPS内核版本 4.4.0-66-generic
通过apt-get upgrade将软件包更新后还是同样的错误

### (2) 路由器型号和固件版本

A：
路由器是基于i7处理器的X86平台软路由

### (3) 详细日志

A：
{standard input}:34056: Warning: end of file not at end of a line; newline inserted
{standard input}:35016: Error: expecting operand after ','; got nothing
{standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
x86_64-openwrt-linux-musl-g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://bugs.openwrt.org/> for instructions.
tools/v8_gypfiles/v8_base_without_compiler.target.mk:590: recipe for target '/home/liberty/Git/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/api/api.o' failed
make[5]: *** [/home/liberty/Git/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/api/api.o] Error 4
rm 43b071ad3d0a8804cbb20c88c8af4d488ab2b411.intermediate 9fbf010d0b2039dc221b45ead9887773b374aa61.intermediate 2d6cc7c117d0df42bddb32d6ca9d1db63b83b05c.intermediate 69b68ef07100d406b929b393370898aca4439081.intermediate
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/liberty/Git/lede/build_dir/target-x86_64_musl/node-v12.15.0'
Makefile:150: recipe for target '/home/liberty/Git/lede/build_dir/target-x86_64_musl/node-v12.15.0/.built' failed
make[3]: *** [/home/liberty/Git/lede/build_dir/target-x86_64_musl/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/liberty/Git/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#30.90#33.06#748.71
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/liberty/Git/lede'
package/Makefile:107: recipe for target '/home/liberty/Git/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/liberty/Git/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/liberty/Git/lede'
/home/liberty/Git/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

AmadeusGhost, 2020-03-09:
这个是因为你的编译机内存不足

hhmaomao, 2020-03-09:
编译起码要有4g内存才稳

gudumuyu, 2020-03-09:
> 这个是因为你的编译机内存不足

请问是怎么判断是因为内存不足引起的呢？

AmadeusGhost, 2020-03-09:
> 请问是怎么判断是因为内存不足引起的呢？

谷歌搜索报错

mjyhj, 2020-03-09:
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib

依赖你都不更新

closed 2020-03-09 by coolsnowwolf