coolsnowwolf/lede #3673: qBittorrent的WebUI端口转发配置未生效导致8080端口被转发到WAN

wspzf, 2020-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：qBittorrent的WebUI端口转发配置在luci里设置的未勾选,保存,
[image]
但qBittorrent v4.1.9 Web 用户界面里的配置是勾选的,手动取消,在luci保存也会再次勾选.
[image]
导致导致8080端口被转发到WAN
[image]

### (2) 路由器型号和固件版本

A：

主机型号 Intel(R) Atom(TM) CPU N2600 @ 1.60GHz : 2 Core 4 Thread
固件版本  OpenWrt R20.3.8 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志

A无

R3pl4c3r, 2020-03-09:
#3679

R3pl4c3r, 2020-03-09:
更改/etc/init.d/qbittorrent的第75行
`'UseUPnP:or("true","false"):true' \`改为`'UseUPnP:or("true","false"):false' \`
注意是UseUPnP而不是UPnP
之后重启qbittorrent即可解决问题

wspzf, 2020-03-09:
> 更改/etc/init.d/qbittorrent的第75行
> `'UseUPnP:or("true","false"):true' \`改为`'UseUPnP:or("true","false"):false' \`
> 注意是UseUPnP而不是UPnP
> 之后重启qbittorrent即可解决问题

感谢!!!,关闭了

closed 2020-03-09 by wspzf