coolsnowwolf/lede #3674: SSR+ 多个订阅链接问题（V2ray）

caicaicai21, 2020-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
如果订阅更新有两条(v2ray的订阅)，订阅更新前如果选择的是订阅连接2的服务器，订阅更新后并不能继续保留所选服务器，但如果更新前是选择订阅连接1的服务器，则可以继续保留所选的服务器。我就觉得很奇怪。
A：

### (2) 路由器型号和固件版本
路由器：x86_64
固件版本：固件内容同步更新至“luci-app-ssr-plus: keep subscribe order"处，本地Ubuntu编译的固件
A：

### (3) 详细日志
2020-03-09 11:12:32 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-09 11:12:31 订阅更新成功
2020-03-09 11:12:31 新增节点数量: 21 删除节点数量: 21
2020-03-09 11:12:31 **当前主服务器节点已被删除，正在自动更换为第一个节点。**
2020-03-09 11:12:31 成功解析节点数量: 26
2020-03-09 11:12:31 成功解析: v2ray 节点, 京港02nf[10倍]
2020-03-09 11:12:31 成功解析: v2ray 节点, 京港01 [10倍]
2020-03-09 11:12:31 成功解析: v2ray 节点, 沪台02nf[10倍]
2020-03-09 11:12:31 成功解析: v2ray 节点, 沪港02nf[10倍]
2020-03-09 11:12:31 成功解析: v2ray 节点, 沪港01 [10倍]
2020-03-09 11:12:31 成功解析: v2ray 节点, 美国 02 nf
2020-03-09 11:12:31 成功解析: v2ray 节点, 美国 01 nf
2020-03-09 11:12:31 成功解析: v2ray 节点, 日本04
2020-03-09 11:12:31 成功解析: v2ray 节点, 日本03
2020-03-09 11:12:31 成功解析: v2ray 节点, 日本02
2020-03-09 11:12:31 成功解析: v2ray 节点, 日本01
2020-03-09 11:12:31 成功解析: v2ray 节点, 新加坡04 NF
2020-03-09 11:12:31 成功解析: v2ray 节点, 新加坡03 NF
2020-03-09 11:12:31 成功解析: v2ray 节点, 新加坡02 NF
2020-03-09 11:12:31 成功解析: v2ray 节点, 新加坡01 NF
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港10
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港09
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港08
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港07
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港06
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港05
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港04
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港03
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港02
2020-03-09 11:12:31 成功解析: v2ray 节点, 香港01
2020-03-09 11:12:31 成功解析: v2ray 节点, 俄罗斯1G 01

2020-03-09 11:12:31 (18.5 KB/s) - written to stdout [9614]

     0K .........                                              18.5K=0.5s

Saving to: 'STDOUT'
Length: unspecified [application/octet-stream]
HTTP request sent, awaiting response... 200 OK
Connecting to sub88.xyz|104.27.148.97|:443... connected.
Resolving sub88.xyz... 104.27.148.97, 104.27.149.97
--2020-03-09 11:12:28--  https://xxx
2020-03-09 11:12:28 成功解析节点数量: 21
2020-03-09 11:12:28 成功解析: v2ray 节点, 日本丨专线丨GMO丨2x
2020-03-09 11:12:28 成功解析: v2ray 节点, 美国丨专线丨DMIT丨2x
2020-03-09 11:12:28 成功解析: v2ray 节点, 美国丨专线丨Cogent GIA丨2x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨专线丨WTT丨2x
2020-03-09 11:12:28 成功解析: v2ray 节点, 台湾丨专线丨Hinet丨2x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨专线丨HKT丨2x
2020-03-09 11:12:28 成功解析: v2ray 节点, 日本丨中继丨GMO-II丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 日本丨中继丨GMO-I丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 美国丨中继丨DMIT丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 美国丨中继丨Cogent GIA丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨中继丨WTT-III丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨中继丨WTT-II丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨中继丨WTT-I丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 台湾丨中继丨Hinet-III丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 台湾丨中继丨Hinet-II丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 台湾丨中继丨Hinet-I丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨中继丨HKT-III丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨中继丨HKT-II丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨中继丨HKT-I丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 台湾丨中继丨bywavesite.github.io丨0.7x
2020-03-09 11:12:28 成功解析: v2ray 节点, 香港丨中继丨TG@bywave丨0.7x

2020-03-09 11:12:28 (88.5 MB/s) - written to stdout [9324]

     0K .........                                              88.5M=0s

Saving to: 'STDOUT'
Length: unspecified [text/html]
HTTP request sent, awaiting response... 200 OK
Connecting to sub.lbbsw.com|47.244.96.61|:443... connected.
Resolving sub.lbbsw.com... 47.244.96.61
--2020-03-09 11:12:28--  https://xxx
2020-03-09 11:12:27 服务正在暂停
A：

caicaicai21, 2020-03-09:
虽然提示说服务器不存在，但实际上是存在的。

coolsnowwolf, 2020-03-09:
没听懂在说什么，先关闭了

closed 2020-03-09 by caicaicai21