coolsnowwolf/lede #3675: 今天编译的版本V2RAY服务器内的SOCKS5无法在外代理TG

tone88, 2020-03-09:
最新代码编译的版本，发现在外面的时候无法使用V2RAY服务器下的SOCKS5给TG做代理。
具体情况：
1、在局域网内的时候TG代理为可用状态，如图2。
2、当出门在外的时候状态就变为不可用了，如图1。这时候如果通过VPN（测试soft vpn和zt VPN都可以）连接回家状态马上变为可用。   后来怀疑是大雕默认取消编译V2RAY服务器后需要加防火墙规则，加结果添加后还是不行。
图中最上面的是新建的SSR+内置的SOCKS5 ,  这个不管是在外还是在局域网内均是可用状态。
[image]
[image]
防火墙添加规则后也不行。
[image]

EnnawYang, 2020-03-09:
经测试正常

tone88, 2020-03-09:
我怀疑是保留配置升级的过，马上重新刷一次测试下

tone88, 2020-03-09:
> 经测试正常使用

全新刷机后问题依旧。

tone88, 2020-03-09:
换了一个端口正常了。  难道是之前使用的端口被什么占用了吗 。端口占用情况用什么情况可以查看？

wspzf, 2020-03-09:
> 换了一个端口正常了。 难道是之前使用的端口被什么占用了吗 。端口占用情况用什么情况可以查看？
我这正常
百度一下嘛，netstat -anp|grep 1080
应该不是端口占用，你在家不是能连吗？

tone88, 2020-03-09:
> > 换了一个端口正常了。 难道是之前使用的端口被什么占用了吗 。端口占用情况用什么情况可以查看？
> > 我这正常
> > 百度一下嘛，netstat -anp|grep 1080
> > 应该不是端口占用，你在家不是能连吗？

V2RAY服务器里的SOCKS我用的12368这个端口。   确实很奇怪，在之前的版本用相同的端口都没问题，就今天升级后才出现这样的问题。  刚刚报着试试的心态换了下端口竟然正常了！

closed 2020-03-09 by coolsnowwolf