coolsnowwolf/lede #3676: 解锁网易云的mini版没了

Orochi-Adde, 2020-03-09:
#3640 # (1) 具体问题

A：解锁网易云的mini版没了

### (2) 路由器型号和固件版本

A：k2p

### (3) 详细日志

A：今天编译更新了一下 发现mini版被移除了 我的路由器性能不好 用mini版效率更高 求加上mini版的解锁网易云 
谢谢

EnnawYang, 2020-03-09:
合并了

guochan2019, 2020-03-09:
你就不看Commits么？依赖都保留了，自已加上不难

Orochi-Adde, 2020-03-09:
> 合并了

合并到那个选项了 是go 还是 nodejs版
还是多了一个新选项

Orochi-Adde, 2020-03-09:
> 你就不看Commits么？依赖都保留了，自已加上不难

太高深了不会啊

mjyhj, 2020-03-09:
> > 合并了
> 
> 合并到那个选项了 是go 还是 nodejs版
> 还是多了一个新选项

2个 都不选就是mini

Orochi-Adde, 2020-03-09:
> > > 合并了
> > 
> > 
> > 合并到那个选项了 是go 还是 nodejs版
> > 还是多了一个新选项
> 
> 2个 都不选就是mini

都不选默认集成mini吗

mjyhj, 2020-03-09:
> > > > 合并了
> > > 
> > > 
> > > 合并到那个选项了 是go 还是 nodejs版
> > > 还是多了一个新选项
> > 
> > 
> > 2个 都不选就是mini
> 
> 都不选默认集成mini吗

怎么选都集成mini，另外2个是附加的。3合1

Orochi-Adde, 2020-03-09:
> > > > > 合并了
> > > > 
> > > > 
> > > > 合并到那个选项了 是go 还是 nodejs版
> > > > 还是多了一个新选项
> > > 
> > > 
> > > 2个 都不选就是mini
> > 
> > 
> > 都不选默认集成mini吗
> 
> 怎么选都集成mini，另外2个是附加的。3合1

懂了 非常感谢

yinzhefuyun, 2020-03-09:
GO的有bug，开代理会无法替换音乐链接，今天早上原作者刚刚更新到0.1.9后修复的https://github.com/cnsilvan/UnblockNeteaseMusic/commit/8dd3f9779fcdaa271414a85b80f43f11300e08d3

closed 2020-03-09 by coolsnowwolf