coolsnowwolf/lede #3682: 请教各位，如何实现URL重定向

404user, 2020-03-09:
是这样的，iOS的Safari 默认的谷歌搜索很坑，是打开google.cn，所以每次搜索都会有一个烦人的cn界面让你重定向到google.com.hk 。
小飞机里是可以把google.cn定向到google.com 。但是不知道软路由行不行，请教一下各位。谢谢

joketans, 2020-03-09:
软路由应该可以改host指向吧，把google.cn 的ip 指到 google.com的ip 试试。我刚用上软路由，不指导能不能改。

404user, 2020-03-09:
[image]
在这里设置了，不过没效果，不知道还有没其他方法。谢谢

> 软路由应该可以改host指向吧，把google.cn 的ip 指到 google.com的ip 试试。我刚用上软路由，不知道能不能改。

hplee0120, 2020-03-09:
手机地区改到非大陆就好了。

joketans, 2020-03-10:
> [image]
> 在这里设置了，不过没效果，不知道还有没其他方法。谢谢 
 提示本地dns要设置到路由地址呢，你设置试试呗

70599, 2020-03-10:
@404user 你在iOS设置里把地区改成美国就行了，默认的是com域名。

closed 2020-07-04 by coolsnowwolf