coolsnowwolf/lede #3687: 最新编译失败  求助

Orochi-Adde, 2020-03-09:
Q：是否搜索了issue (使用 "x" 选择)
* [ *] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败 怎么弄都弄不来 已经试过 
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler
和
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib
和
sudo apt-get install gcc-multilib g++-multilib
安装的时候都显示是最新的 无需安装
adde@DESKTOP-B4PR0NT:~/lede$ sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
Note, selecting 'libpython3.5-minimal' for regex 'python3.5'
Note, selecting 'node-uglify' instead of 'uglifyjs'
Note, selecting 'git' instead of 'git-core'
Note, selecting 'upx-ucl' instead of 'upx'
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.15.1-3ubuntu2).
build-essential is already the newest version (12.4ubuntu1).
device-tree-compiler is already the newest version (1.4.5-3).
flex is already the newest version (2.6.4-6).
gawk is already the newest version (1:4.1.4+dfsg-1build1).
libc6-dev-i386 is already the newest version (2.27-3ubuntu1).
libtool is already the newest version (2.4.6-2).
unzip is already the newest version (6.0-21ubuntu1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-0ubuntu2).
asciidoc is already the newest version (8.6.10-2).
msmtp is already the newest version (1.6.6-1).
node-uglify is already the newest version (2.8.29-3).
p7zip is already the newest version (16.02+dfsg-6).
p7zip-full is already the newest version (16.02+dfsg-6).
subversion is already the newest version (1.9.7-4ubuntu1).
texinfo is already the newest version (6.5.0.dfsg.1-2).
upx-ucl is already the newest version (3.94-4).
xmlto is already the newest version (0.0.28-2).
autopoint is already the newest version (0.19.8.1-6ubuntu0.3).
binutils is already the newest version (2.30-21ubuntu1~18.04.2).
bzip2 is already the newest version (1.0.6-8.1ubuntu0.2).
g++-multilib is already the newest version (4:7.4.0-1ubuntu2.3).
gcc-multilib is already the newest version (4:7.4.0-1ubuntu2.3).
gettext is already the newest version (0.19.8.1-6ubuntu0.3).
git is already the newest version (1:2.17.1-1ubuntu0.5).
lib32gcc1 is already the newest version (1:8.3.0-6ubuntu1~18.04.1).
libelf-dev is already the newest version (0.170-0.4ubuntu0.1).
libglib2.0-dev is already the newest version (2.56.4-0ubuntu0.18.04.4).
libncurses5-dev is already the newest version (6.1-1ubuntu1.18.04).
libssl-dev is already the newest version (1.1.1-1ubuntu2.1~18.04.5).
patch is already the newest version (2.7.6-2ubuntu1.1).
qemu-utils is already the newest version (1:2.11+dfsg-1ubuntu7.23).
0 upgraded, 0 newly installed, 0 to remove and 131 not upgraded.

### (2) 路由器型号和固件版本

A：k2p

### (3) 详细日志

A：这是编译错误日志
tools/v8_gypfiles/v8_base_without_compiler.target.mk:26: warning: overriding recipe for target '/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Runtime.h'tools/v8_gypfiles/v8_base_without_compiler.host.mk:26: warning: ignoring old recipe for target '/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Runtime.h'tools/v8_gypfiles/v8_base_without_compiler.target.mk:26: warning: overriding recipe for target '/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
tools/v8_gypfiles/v8_base_without_compiler.host.mk:26: warning: ignoring old recipe for target '/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
  LD_LIBRARY_PATH=/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/lib.host:/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated; python ../../deps/v8/tools/run.py "/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/bytecode_builtins_list_generator" "/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h"
Traceback (most recent call last):
  File "../../deps/v8/tools/run.py", line 12, in <module>
    sys.exit(subprocess.call(sys.argv[1:]))
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
tools/v8_gypfiles/generate_bytecode_builtins_list.target.mk:13: recipe for target '/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h' failed
make[5]: *** [/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h] Error 1
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0'
Makefile:150: recipe for target '/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built' failed
make[3]: *** [/home/adde/lede/build_dir/target-mipsel_24kc_musl/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/adde/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#0.17#2.17#2.78
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/adde/lede'
package/Makefile:107: recipe for target '/home/adde/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/adde/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/adde/lede'
/home/adde/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

Orochi-Adde, 2020-03-09:
我是win10子系统下的ubuntu18.04

closed 2020-03-09 by coolsnowwolf

sundog315, 2020-03-12:
遇到了同样的问题

klever1988, 2020-03-19:
@sundog315 @Orochi-Adde 
把第一条apt-get install命令里的python3.5改成python3
试试看