coolsnowwolf/lede #3692: wan口或者lan口dns故障

adamgonglj, 2020-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

lan口dns故障。

ssrplus，可以访问外网；

但是无法访问国内网站，经常提示正在连接中，经检测dns设置正确，尝试过将dns设置成为上级路由，尝试设置成114.114.114.114。

均存在国内网站无法访问，国外网站可以访问的现象（ssr+中的dns设置的是8.8.8.8）

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
固件版本，根据最新的源码编译的。
固件版本号：4.19.106

A：

### (3) 详细日志

这是系统日志。
Mon Mar  9 20:51:09 2020 user.warn ddns-scripts[3913]: myddns_ipv4: Service section disabled! - TERMINATE
Mon Mar  9 20:51:09 2020 user.warn ddns-scripts[3913]: myddns_ipv4: PID '3913' exit WITH ERROR '1' at 2020-03-09 20:51
Mon Mar  9 20:51:09 2020 user.info mwan3track[3693]: Check (ping) failed for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:51:15 2020 user.info mwan3track[3693]: Lost 1 ping(s) on interface wan (eth1)
Mon Mar  9 20:51:22 2020 user.info mwan3track[3693]: Check (ping) failed for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:51:29 2020 user.info mwan3track[3693]: Check (ping) failed for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:51:34 2020 user.info mwan3track[3693]: Lost 2 ping(s) on interface wan (eth1)
Mon Mar  9 20:51:46 2020 user.info mwan3track[3693]: Check (ping) failed for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:51:53 2020 user.info mwan3track[3693]: Check (ping) failed for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:00 2020 user.info mwan3track[3693]: Check (ping) failed for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:00 2020 user.notice mwan3track[3693]: Interface wan (eth1) is offline
Mon Mar  9 20:52:00 2020 user.notice mwan3[5452]: Execute ifdown event on interface wan (eth1)
Mon Mar  9 20:52:00 2020 user.info mwan3[5452]: connection tracking not flushed on interface wan (eth1) ifdown
Mon Mar  9 20:52:01 2020 user.info mwan3track[3693]: Detect ifdown event on interface wan (eth1)
Mon Mar  9 20:52:08 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:22 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:22 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:22 2020 user.info mwan3track[3693]: Lost 6 ping(s) on interface wan (eth1)
Mon Mar  9 20:52:27 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:27 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:32 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:32 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:37 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:37 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:42 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:42 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:47 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:47 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:52:52 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:52:53 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:53:00 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:53:05 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:53:12 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:53:12 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:53:12 2020 user.info mwan3track[3693]: Lost 2 ping(s) on interface wan (eth1)
Mon Mar  9 20:53:17 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:53:24 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:53:49 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.114.114" on interface wan (eth1)
Mon Mar  9 20:53:49 2020 user.info mwan3track[3693]: Check (ping) success for target "114.114.115.115" on interface wan (eth1)
Mon Mar  9 20:53:49 2020 user.info mwan3track[3693]: Lost 6 ping(s) on interface wan (eth1)
[... 1315 lines cut ...]
[ 4823.258814] br-wan: port 1(eth1) entered forwarding state
[ 4959.854425] br-wan: port 1(eth1) entered disabled state
[ 4959.862235] device eth1 left promiscuous mode
[ 4959.866961] br-wan: port 1(eth1) entered disabled state
[ 4960.025510] br-wan: port 1(eth1) entered blocking state
[ 4960.030942] br-wan: port 1(eth1) entered disabled state
[ 4960.036755] device eth1 entered promiscuous mode
[ 4960.042570] br-wan: port 1(eth1) entered blocking state
[ 4960.047993] br-wan: port 1(eth1) entered forwarding state
[ 5822.772882] br-wan: port 1(eth1) entered disabled state
[ 5822.781784] device eth1 left promiscuous mode
[ 5822.786500] br-wan: port 1(eth1) entered disabled state
[ 5823.018129] RTL8211E Gigabit Ethernet r8169-300:00: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=r8169-300:00, irq=IGNORE)
[ 5823.266541] r8169 0000:03:00.0 eth1: Link is Down
[ 5826.276209] r8169 0000:03:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[ 6284.181128] r8169 0000:03:00.0 eth1: Link is Down
[ 6289.667654] r8169 0000:03:00.0 eth1: Link is Up - 100Mbps/Full - flow control off
[ 6326.189794] r8169 0000:03:00.0 eth1: Link is Down
[ 6332.848710] r8169 0000:03:00.0 eth1: Link is Up - 100Mbps/Full - flow control off
[86041.900330] r8169 0000:03:00.0 eth1: Link is Down
[86061.329543] r8169 0000:03:00.0 eth1: Link is Up - 1Gbps/Full - flow control off

其他人的反馈情况：

https://www.chiphell.com/thread-1730354-1-1.html

AmadeusGhost, 2020-03-09:
卸载掉mwan3即可

adamgonglj, 2020-03-09:
真的？

adamgonglj, 2020-03-09:
正在卸载，如果仍然有问题，我再反馈。

adamgonglj, 2020-03-09:
问题解决。

closed 2020-03-09 by adamgonglj