coolsnowwolf/lede #3695: Tidal music 在ssr plus下的速度问题

baygixu, 2020-03-09:
我使用的的n1盒子做旁路由，节点是v2ray tls+ws，开大陆模式，tcp和udp都用这个节点。现在碰到一个奇怪的问题是，手机上tidal music的速度非常慢，但用4g不翻墙或手机v2ray客户端翻墙，tidal音乐的速度都非常快，就感觉这个app连接n1的时候不知道有没有连接代理，怀疑是不是udp或者ipv6的原因，还是有别的可能导致？请问有办法路由器翻墙的时候，针对手机的某个app做强制代理或不代理吗？感谢各位大神解答，谢谢

hhmaomao, 2020-03-09:
考虑一下域名强制代理，如果指定app安卓手机上的v2rayn可以实现，但软路由不知道手机上有什么app啊

baygixu, 2020-03-09:
> 考虑一下域名强制代理，如果指定app安卓手机上的v2rayn可以实现，但软路由不知道手机上有什么app啊

是啊，我就愁这个，软路由不知道怎么强制代理某个安卓app，而且也不知道这个app的具体域名。更奇怪的就算没走代理，也不该这么慢啊？而且好像网页版挺快的

R3pl4c3r, 2020-03-10:
> > 考虑一下域名强制代理，如果指定app安卓手机上的v2rayn可以实现，但软路由不知道手机上有什么app啊
> 
> 是啊，我就愁这个，软路由不知道怎么强制代理某个安卓app，而且也不知道这个app的具体域名。更奇怪的就算没走代理，也不该这么慢啊？而且好像网页版挺快的

你可以用抓包工具（比如fiddler）看一下它访问了哪些域名，然后对这些域名强制走代理即可

baygixu, 2020-03-10:
> > > 考虑一下域名强制代理，如果指定app安卓手机上的v2rayn可以实现，但软路由不知道手机上有什么app啊
> > 
> > 
> > 是啊，我就愁这个，软路由不知道怎么强制代理某个安卓app，而且也不知道这个app的具体域名。更奇怪的就算没走代理，也不该这么慢啊？而且好像网页版挺快的
> 
> 你可以用抓包工具（比如fiddler）看一下它访问了哪些域名，然后对这些域名强制走代理即可

看了下，主要是有个UDP的239.255.255.250:1900的request，不知道是不是和这个有关啊，如果是的话，应该怎么设置呢？

abclai, 2020-03-10:
对手机开启游戏模式吧！plus好像只能TCP，你这个用到UDP的话要对手机开启游戏模式 这样才能代理UDP

baygixu, 2020-03-10:
> 对手机开启游戏模式吧！plus好像只能TCP，你这个用到UDP的话要对手机开启游戏模式 这样才能代理UDP

感谢回复，我现在在ssr plus配置的首页上，游戏中继模式udp是选了同一个v2ray的节点，不知道是指这样设置吗？还是有别的UDP的设置方式，尤其是针对手机开UDP？谢谢

MagicLentil, 2020-03-11:
我也有同样的问题，但是我是mac

baygixu, 2020-03-11:
> 我也有同样的问题，但是我是mac

果然有人有一样的问题啦，如果方便能帮忙看下手机上的情况吗？安卓或iPhone都可以啊，谢谢

hhmaomao, 2020-03-11:
ssr plus不是有个：游戏模式UDP中继服务器，选择与全局服务器相同理论上就可以了

baygixu, 2020-03-11:
> ssr plus不是有个：游戏模式UDP中继服务器，选择与全局服务器相同理论上就可以了
我试过设置成和主服务器一个节点或和全局服务器相同，都不行

lizhi66662000, 2020-03-12:
既然是固定的环境，为什么不能直接打个包让大家下载，非要在网上去下载，花大量的时间，真不理解。而且下载的效果也不好。

baygixu, 2020-03-12:
> 既然是固定的环境，为什么不能直接打个包让大家下载，非要在网上去下载，花大量的时间，真不理解。而且下载的效果也不好。

什么包啊，不太明白是指哪个？

baygixu, 2020-03-14:
其实不止tidal，Facebook lite和messenger lite也有这个问题，比如Facebook lite卡在开屏加载的界面上，但正常的Facebook app就没问题，网页版也OK。另外如果用手机端v2ray的客户端，Facebook lite就正常了

MagicLentil, 2020-03-16:
我现在换了Surge 4，一切都解决了，软件的问题，之前用的clashx、v2rayu、v2rayx都不行。

baygixu, 2020-03-16:
> 我现在换了Surge 4，一切都解决了，软件的问题，之前用的clashx、v2rayu、v2rayx都不行。

请问是指手机客户端吗？手机客户端我都可以的，但走路由器科学就不行，请问你路由器端用的是？

baygixu, 2020-03-17:
这个问题奇怪的在于，节点肯定是好的（通过手机v2ray客户端和其他需翻墙的网站证明），手机和tidal的app也是好的，路由器也是好的（可以科学上网），但连起来就不行。今天用httpCanary试图在手机上抓包的时候发现，连接了家里的wifi（可翻墙），tidal的速度居然很快，但一断开httpCanary（还是连着家里wifi）,马上就不行了。这到底是怎么回事呀

baygixu, 2020-03-20:
现在看来不是科学上网的问题，尝试手机直连不科学的主路由，速度正常，然后关了旁路由科学上网的插件，连上旁路由，速度就很慢，所以应该喝科学上网无关。而且速度不是一直慢，而是大部分时间很慢，偶尔快一下，所以是旁路由的设置有什么问题吗？

hcym, 2020-03-20:
单口路由就看hanwckf的博客方法，什么强制dns劫持网关的都是歪门邪道，用他的方法使用体验和普通硬路由一模一样一模一样

baygixu, 2020-03-21:
> 单口路由就看hanwckf的博客方法，什么强制dns劫持网关的都是歪门邪道，用他的方法使用体验和普通硬路由一模一样一模一样

能麻烦分享下博客的链接吗？谢谢啊

closed 2020-07-04 by coolsnowwolf