coolsnowwolf/lede #3697: Windows子系统编译报错

Troyshen94, 2020-03-10:
## 1.关于你要提交的问题
编译系统: Windows子系统Ubuntu 18.04 LTS
Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译提示

find /home/troyshen/lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1583753131" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2019.522.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/troyshen/lede'
package/Makefile:108: recipe for target '/home/troyshen/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/troyshen/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/troyshen/lede'
/home/troyshen/lede/include/toplevel.mk:225: recipe for target 'world' failed

求助各位大佬，发生了什么 0-0

EnnawYang, 2020-03-10:
请搜索之前的issue谢谢

closed 2020-03-10 by Troyshen94

lengyuqingqiu, 2020-03-10:
放弃子系统吧 我用子系统就没编译成功过。各种各样的问题，老实用虚拟机吧。