coolsnowwolf/lede #3698: ssr plus+ 中的SOCKS5 代理服务端的疑问

wsndshx, 2020-03-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：梯子为v2ray，在以GFW模式运行时，开启SOCKS5 代理服务端，chrome浏览器通过插件Proxy SwitchyOmega指向路由器开启的SOCKS5服务器。但此时无法访问外面的世界，查询得到的ip也为本地运营商提供的ip。而当浏览器插件为直连模式（透明代理）时可以正常访问外面的世界。
[image]
[image]
[image]
[image]

### (2) 路由器型号和固件版本

A：Newifi-D2 | OpenWrt R20.3.8 / LuCI Master (git-20.068.29105-b2cf61d)

## 3. 疑问
A：ssr plus+ 中的SOCKS5代理服务器是提供直连的吗，还是说是提供全局代理只不过不支持v2ray的梯子？

wsndshx, 2020-03-10:
找到了个[去年的帖子](https://github.com/coolsnowwolf/lede/issues/1296)，发现里面说SOCKS5只支持ssr代理……emmmmmm可是我记得之前这样用是可以访问外网的……

EnnawYang, 2020-03-10:
煋

wsndshx, 2020-03-10:
> 煋

emmm那我可能记错了吧（

closed 2020-03-10 by wsndshx

guochan2019, 2020-03-10:
刚才试了一下，的确如此，不只是V2ray，Troian也是，换成绕过中国大陆模式可以，GFW不行